orocos/orocos_kinematics_dynamics #42: PyKDL: specifying a root name when intializing a Tree causes segfault

jf---, 2015-01-15:
```python
from PyKDL import Tree
ohno = Tree(root_name="carrot")
# kaboom
```
Segmentation fault: 11

smits, 2015-01-15:
Can you provide a backtrace?

R.

jf---, 2015-01-15:
Sorry Ruben, since this leads to a segfault I can't really do so...
Or do you mean a GDB trace?

jf---, 2015-01-15:
Btw, I noticed that trying to access a non-existing segment, results into segfault.
I guess you'd expect a ValueError to be raised

```python
chain.getSegment( chain.getNrOfSegments() + 1 )
# poooof -> segfault
```

smits, 2015-06-09:
I would indeed need a gdb trace. And there is indeed no index checking in the getSegment function: https://github.com/orocos/orocos_kinematics_dynamics/blob/master/orocos_kdl/src/chain.cpp#L68-L71

zchen24, 2018-05-16:
@jf--- @smits #145 should fix this.

MatthijsBurgh, 2020-04-18:
@jf--- please close this issue, if fixed

closed 2020-04-18 by jf---