orocos/orocos_kinematics_dynamics #46: chainiksolvervel_pinv_nso math implementation incorrect

rethink-forrest, 2015-02-12:
I am working on a 7 DOF arm. With testing, chainiksolvervel_pinv_nso gives the exact same answer of chainiksolvervel_pinv every time.

Looking at the code, the nullspace math is not implemented properly on lines 106 - 123. There is no documentation for the attempted approach, and the reference paper from 1977 uses a different optimization without SVD decomposition.

As far as I can guess, the attempted technique is

t = weights * (q_opt - q_in)

q'_out += alpha * V_c * V_n^T * t

where V_c is the first 6 columns of V, and V_n are the remaining columns. Is this correct, and is there a paper or other references for this method?

If I know the desired method, I should be able to submit a patch quickly.
 
Other issues:
Line 110: The index i should start at jac.rows() and not jac.rows()+1. For a 7 dof arm, jac is 6x7, and this loop will never be run as written.  Same error is repeated in line 119.

Line 120: attempts to read tmp2 at index jac.rows()+1, but tmp2 only has jac.columns()-jac.rows() elements.

smits, 2015-02-13:
> I am working on a 7 DOF arm. With testing, chainiksolvervel_pinv_nso gives
> the exact same answer of chainiksolvervel_pinv every time.
>
> Looking at the code, the nullspace math is not implemented [...]

The paper does not explain how it calculates the generalized inverse of J,
this code uses an SVD to do that

> As far as I can guess, the attempted technique is
>
> t = weights * (q_opt - q_in)
>
> q'_out += alpha * V_c * V_n^T * t

According to the paper it should be q'_out += (I_n - G_2*J)*alpha*t

with G_2 = J^t(J*J^t)^-1 (aka pseudo inverse)

and (I_n-G_2*J) the projection of alpha*t into the null-space.

I have not looked into the math yet if the current code results in the same.

> where V_c is the first 6 columns of V, and V_n are the remaining columns.
> Is this correct, and is there a paper or other references for this method?
>
> If I know the desired method, I should be able to submit a patch quickly.
Other issues:
Line 110: The index i should start at jac.rows() and not jac.rows()+1. For
> a 7 dof arm, jac is 6x7, and this loop will never be run as written. Same
> error is repeated in line 119.

This indeed seems wrong.

> Line 120: attempts to read tmp2 at index jac.rows()+1, but tmp2 only has
> jac.columns()-jac.rows() elements.
This is odd.

Ruben

rethink-forrest, 2015-02-13:
I think the given equations are missing an additional Jacobian which projects the cost function into the joint space.  Here is how I've worked out the math.

Given the cost function g, and the current joints q, desired joints qd, and weights:

t = g(q) = 1/2 * Sum( wt_i * (q_i - qd_i)^2 )

The jacobian Jc is:

t_dot = Jc(q) * q_dot
Jc = dt/dq = w_j * (q_i - qd_i)   [1x7 vector]

The pseudo inverse (^-1) is

Jc^-1 = w_j * (q_i - qd_i) / A   [7x1 vector]
A = Sum( w_i^2 * (q_i - qd_i)^2 )

In the current code, tmp is the inverse Jacobian without the needed normalization term A. We can set the change as the step needed to remove the error times a value alpha:

t_dot = -2 * alpha * t

When we put it together and project into the nullspace, the final result is

q'_out += (I_n - J^-1*J) * Jc^-1 * (-2*alpha*t)

Given the extra matrix multiplications and the fact that trying to do it via nested for loops are prone to error, I'm probably going to convert everything into Eigen matrices and vectors as was done in chainiksolvervel_wdls.

rethink-forrest, 2015-02-13:
I have a working version with the above method. It works in simulation, and I'll test it on an actual robot next week.

smits, 2015-04-09:
Close by #50

closed 2015-04-09 by smits