orocos/orocos_kinematics_dynamics #54: inertiatest fails on 32bit architectures

morxa, 2015-04-30:
I'm currently trying to put KDL into a Fedora package, but the inertiatest consistently fails on 32bit architectures. As an example, see [this build log](https://kojipkgs.fedoraproject.org//work/tasks/7845/9607845/build.log). 

**How to reproduce:**

1. Build on a 32bit architecture
1. Run the tests

**Expected behavior**
All tests should terminate successfully.

**Actual behavior**
The intertia test fails, all other tests complete successfully.

Unfortunately, I don't have access to a 32bit computer myself, so I can only point to the Fedora build log. But I assume the behavior is the same. If required, I can set up a 32bit virtual machine to do further investigation.

morxa, 2015-05-20:
Is there any progress on this? If you need more information, please let me know.

smits, 2015-06-23:
Could you provide the contents of the resulting inertiatest-result.xml file, it will tell us exactly which test failed.

morxa, 2015-06-23:
This is the `inertiatest-result.xml` of the build above:

```
<?xml version="1.0" encoding='ISO-8859-1' standalone='yes' ?>
<TestRun>
  <FailedTests>
    <FailedTest id="3">
      <Name>InertiaTest::TestArticulatedBodyInertia</Name>
      <FailureType>Assertion</FailureType>
      <Location>
        <File>/home/robocup/rpmbuild/SOURCES/orocos_kinematics_dynamics-2aa76640f0a1c5ac57946c20e844372578b55743/orocos_kdl/tests/inertiatest.cpp</File>
        <Line>176</Line>
      </Location>
      <Message>equality assertion failed
- Expected:   1.9436e-17     0.152465    -0.673686
   -0.152465  4.68917e-18    -0.592164
    0.673686     0.592164 -3.38542e-17
- Actual  :  2.13418e-17     0.152465    -0.673686
   -0.152465  1.52059e-17    -0.592164
    0.673686     0.592164 -1.84585e-17
</Message>
    </FailedTest>
  </FailedTests>
  <SuccessfulTests>
    <Test id="1">
      <Name>InertiaTest::TestRotationalInertia</Name>
    </Test>
    <Test id="2">
      <Name>InertiaTest::TestRigidBodyInertia</Name>
    </Test>
  </SuccessfulTests>
  <Statistics>
    <Tests>3</Tests>
    <FailuresTotal>1</FailuresTotal>
    <Errors>0</Errors>
    <Failures>1</Failures>
  </Statistics>
</TestRun>
```

Interestingly, I've just tried to reproduce it in a virtual machine. Now, `velocityprofiletest` fails, while the `inertiatest` runs successfully. Here is `velocityprofiletest-result.xml`:

```
<?xml version="1.0" encoding='ISO-8859-1' standalone='yes' ?>
<TestRun>
  <FailedTests>
    <FailedTest id="7">
      <Name>VelocityProfileTest::TestTrapHalf_SetDuration_Start</Name>
      <FailureType>Assertion</FailureType>
      <Location>
        <File>/home/till/rpmbuild/SOURCES/orocos_kinematics_dynamics/orocos_kdl/tests/velocityprofiletest.cpp</File>
        <Line>259</Line>
      </Location>
      <Message>equality assertion failed
- Expected: 0
- Actual  : 1.11022302462516e-16
</Message>
    </FailedTest>
    <FailedTest id="8">
      <Name>VelocityProfileTest::TestTrapHalf_SetDuration_End</Name>
      <FailureType>Assertion</FailureType>
      <Location>
        <File>/home/till/rpmbuild/SOURCES/orocos_kinematics_dynamics/orocos_kdl/tests/velocityprofiletest.cpp</File>
        <Line>311</Line>
      </Location>
      <Message>equality assertion failed
- Expected: 0
- Actual  : -1.11022302462516e-16
</Message>
    </FailedTest>
  </FailedTests>
  <SuccessfulTests>
    <Test id="1">
      <Name>VelocityProfileTest::TestTrap_MaxVelocity1</Name>
    </Test>
    <Test id="2">
      <Name>VelocityProfileTest::TestTrap_MaxVelocity2</Name>
    </Test>
    <Test id="3">
      <Name>VelocityProfileTest::TestTrap_MaxVelocity3</Name>
    </Test>
    <Test id="4">
      <Name>VelocityProfileTest::TestTrap_SetDuration1</Name>
    </Test>
    <Test id="5">
      <Name>VelocityProfileTest::TestTrapHalf_SetProfile_Start</Name>
    </Test>
    <Test id="6">
      <Name>VelocityProfileTest::TestTrapHalf_SetProfile_End</Name>
    </Test>
  </SuccessfulTests>
  <Statistics>
    <Tests>8</Tests>
    <FailuresTotal>2</FailuresTotal>
    <Errors>0</Errors>
    <Failures>2</Failures>
  </Statistics>
</TestRun>
```

smits, 2015-06-26:
I could reproduce this in a vm as well.

closed 2015-06-26 by smits

morxa, 2015-06-27:
Thanks for fixing this!
FYI: I've applied the patch to the last release (1.3.0) and I'm currently pushing updates to Fedora, so there will be a i686 package soon.

commit 96120fdd (2015-06-26): tests: be more relaxed on calculated double values. Fixes #54
commit 0ecfbf76 (2016-07-06): tests: be more relaxed on calculated double values. Fixes #54