orocos/orocos_kinematics_dynamics #57: JntArray::Equal does not always work

pbeeson, 2015-07-10:
I noticed that doing a KDL::Equal(q_out,KDL:JntArray(q_out.rows()),1000) always returns false.  That's a HUGE epsilon, and it's false.  I tracked the down to a change in KDL, where the hand-coded equal that looped through the array was replaced by the Eigen isApprox() function.  Apparently, is Approx does not work when either of the vectors is very close to zero.  This seems bad/wrong.

pbeeson, 2015-08-12:
This is a pretty large and important bug.  I would have hoped someone would have addressed this by now.

smits, 2015-08-12:
I'll add a unittest for this specific case and see if your proposal fixes that specific issue.

pbeeson, 2015-12-10:
For some reason, the HEAD for the KDL source code still uses isApprox() for a JntArray::Equal(), which makes Equal() return False almost always for two equally sized joint arrays (Because of the size of the array).  This really should be properly fixed.

MatthijsBurgh, 2020-06-15:
@pbeeson I have tested this in PyKDL. I eexperienced the problem you describe. I have create a PR, #262, which gets rid of `isApprox` entirely.

PR #262 (2020-06-30): Don't use isApprox, but isZero
As talked in #57.

Fixes #57
closed 2020-06-30 by MatthijsBurgh