orocos/orocos_kinematics_dynamics #59: Python bindings fail to build with python 3

smits, 2015-07-16:
```
[ 11%] Building CXX object CMakeFiles/python_module_PyKDL.dir/PyKDL/sipPyKDLpart7.o
/home/allenh1/catkin_ws/src/orocos_kinematics_dynamics/python_orocos_kdl/PyKDL/std_string.sip: In function ‘int convertTo_std_string(PyObject*, void**, int*, PyObject*)’:
/home/allenh1/catkin_ws/src/orocos_kinematics_dynamics/python_orocos_kdl/PyKDL/std_string.sip:41:37: error: ‘PyString_Check’ was not declared in this scope
         return (PyString_Check(sipPy) || PyUnicode_Check(sipPy));
                                     ^
/home/allenh1/catkin_ws/src/orocos_kinematics_dynamics/python_orocos_kdl/PyKDL/std_string.sip:48:58: error: ‘PyString_AS_STRING’ was not declared in this scope
         *sipCppPtr = new std::string(PyString_AS_STRING(s));
                                                          ^
/home/allenh1/catkin_ws/src/orocos_kinematics_dynamics/python_orocos_kdl/PyKDL/std_string.sip:52:30: error: ‘PyString_Check’ was not declared in this scope
      if (PyString_Check(sipPy)) {
                              ^
/home/allenh1/catkin_ws/src/orocos_kinematics_dynamics/python_orocos_kdl/PyKDL/std_string.sip:53:62: error: ‘PyString_AS_STRING’ was not declared in this scope
         *sipCppPtr = new std::string(PyString_AS_STRING(sipPy));
                                                              ^
/home/allenh1/catkin_ws/src/orocos_kinematics_dynamics/python_orocos_kdl/PyKDL/std_string.sip: In function ‘PyObject* convertFrom_std_string(void*, PyObject*)’:
/home/allenh1/catkin_ws/src/orocos_kinematics_dynamics/python_orocos_kdl/PyKDL/std_string.sip:30:56: error: ‘PyString_FromString’ was not declared in this scope
         newstring = PyString_FromString(sipCpp->c_str());
                                                        ^
CMakeFiles/python_module_PyKDL.dir/build.make:243: recipe for target 'CMakeFiles/python_module_PyKDL.dir/PyKDL/sipPyKDLpart7.o' failed
```

triffid, 2016-02-12:
Just ran into this, solution from http://answers.ros.org/question/212800/problems-building-from-source/ doesn't seem to make any difference..

is there anything I can do in the meantime to work around this?

humm, 2016-03-25:
@triffid: This seems fixed in the master branch (the fix was applied a year ago). From the workspace dir, you have to:

1. replace the content of `/src/orocos_kinematics_dynamics/python_orocos_kdl/PyKDL/std_string.sip` by the one of https://raw.githubusercontent.com/orocos/orocos_kinematics_dynamics/master/python_orocos_kdl/PyKDL/std_string.sip (check that the file contains `#if PY_MAJOR_VERSION < 3` somewhere).
2. **delete** the `build/python_orocos_kdl/` directory. If you don't, the changes to `std_string.sip` won't be taken into account.
3. relaunch the catkin config/install. 

That did it for me on jade, with Python 3.5.1.

smits, 2016-07-06:
Fixed by 33400d4adaf which is part of v1.3.1

closed 2016-07-06 by smits

jacknlliu, 2018-03-02:
I met this issue again.

```py
from urdf_parser_py.urdf import URDF
import PyKDL as kdl
robot = URDF.from_parameter_server()
root = robot.get_root()
tree=kdl.Tree(root)
```
then output is
```
python3: /home/ros/data/DRL_Platforms/catkin_ws_for_kdl/src/orocos_kinematics_dynamics/python_orocos_kdl/PyKDL/std_string.sip:52: int convertTo_std_string(PyObject*, void**, int*, PyObject*): Assertion `PyUnicode_Check(s)' failed.
Aborted (core dumped)
```

I build PyKDL with python3 following this command
```sh
catkin build  -DPYTHON_VERSION=3.5 -DPYTHON_EXECUTABLE=/usr/bin/python3.5
```

My system information
```
python 3.5.1
orocos/orocos_kinematics_dynamics: latest of this repo
catkin: 0.4.4
Ubuntu 16.04
ROS kinetic
```