orocos/orocos_kinematics_dynamics #67: Error in CMakeLists.txt of orocos/orocos_kinematics_dynamics

korken89, 2016-01-19:
Hi,

I have been hunting a compile bug for this package and finally found it.
See entire search here: https://aur.archlinux.org/packages/ros-indigo-orocos-kdl/#news

The problem is that the `Eigen_INCLUDE_DIR(S)` is different on different systems, and you are using `Eigen_INCLUDE_DIR` to find the include.
However, the standard from the FindEigen.cmake files is `Eigen_INCLUDE_DIRS` <- Note the S on the end.

To fix this, check is `Eigen_INCLUDE_DIR` is empty and then use `Eigen_INCLUDE_DIRS` instead if it is.

I am referring to this file: `orocos_kinematics_dynamics/orocos_kdl/CMakeLists.txt`
Lines: 50-51

If you do not have time I can fix a patch for it.

Thanks!

traversaro, 2016-01-19:
>  However, the standard from the FindEigen.cmake files is Eigen_INCLUDE_DIRS <- Note the S on the end. 

What do you mean by "standard" ? 
Given that upstream Eigen distributes both a `FindEigen3.cmake` and a `Eigen3Conflg.cmake` file and kdl does not support anymore Eigen2, I would suggest fixing the CMakeLists.txt by directly searching `find_package(Eigen3)` in place of `find_package(Eigen 3)` and populating `Eigen_INCLUDE_DIR` from `EIGEN3_INCLUDE_DIR`. 
See https://github.com/ros/cmake_modules/issues/25 .

korken89, 2016-01-19:
That is a better solution, nice!

bchretien, 2016-01-20:
`cmake_modules` should also be added to `package.xml`.

MatthijsBurgh, 2020-05-06:
outdated, please open a new issue if still affected by an Eigen issue.

closed 2020-05-06 by MatthijsBurgh