orocos/orocos_kinematics_dynamics #74: PyKDL Tree - AttributeError

brian-kf, 2016-02-05:
Hi,

I have a problem with the PyKDL module more specifically with the Tree/TreeElement class. 

> import PyKDL as kdl
> kdl.Tree.addSegment
<built-in function addSegment>

> kdl.Tree.getRootSegment
AttributeError: type object 'Tree' has no attribute 'getRootSegment'
> kdl.TreeElement
AttributeError: 'module' object has no attribute 'TreeElement'

I can only access to the addSegment, getChain, getNrOfJoints, and getNrOfSegments methods of the Tree class. I can NOT access to the other methods (getSegments, getRootSegment, getSegment, addTree, addChain, etc), and also the TreeElement class.

I tested it on my MacOSX 10.9.4 and on Ubuntu 14.04, and I have the same problem.

Do you know how to fix it?

Thanks

smits, 2017-02-20:
This functionality is not exported in https://github.com/orocos/orocos_kinematics_dynamics/blob/master/python_orocos_kdl/PyKDL/kinfam.sip#L151-L166 which is why it is not possible to access these functions.  I welcome a PR that adds these.

MatthijsBurgh, 2020-05-04:
Outdated

closed 2020-05-04 by MatthijsBurgh