orocos/orocos_kinematics_dynamics #77: ChainIkSolverVel_wdls Fails on Assert

trhermans, 2016-02-11:
I recently received the same assert as this post: http://www.orocos.org/forum/orocos/orocos-users/position-only-ik:

python: /usr/include/eigen3/Eigen/src/Core/Block.h:141: Eigen::Block<XprType, BlockRows, BlockCols, InnerPanel>::Block(XprType&, Eigen::Block<XprType, BlockRows, BlockCols, InnerPanel>::Index, Eigen::Block<XprType, BlockRows, BlockCols, InnerPanel>::Index, Eigen::Block<XprType, BlockRows, BlockCols, InnerPanel>::Index, Eigen::Block<XprType, BlockRows, BlockCols, InnerPanel>::Index) [with XprType = Eigen::Matrix<double, -1, -1>; int BlockRows = -1; int BlockCols = -1; bool InnerPanel = false; Eigen::Block<XprType, BlockRows, BlockCols, InnerPanel>::Index = long int]: Assertion `a_startRow >= 0 && blockRows >= 0 && a_startRow <= xpr.rows() - blockRows && a_startCol >= 0 && blockCols >= 0 && a_startCol <= xpr.cols() - blockCols' failed.
Aborted (core dumped)

I signed up for the forum but still have not been granted login rights to post there. Thus I figured I'd email you before trying to hack around on the solution myself. I also couldn't find any commits on github relating to this issue, so I imagine it is still active.

I'm solving ChainIkSolverVel_wdls through ChainIkSolverPos_NR_JL() via Python in ros-indigo installed on Ubuntu 14.04.

Eigen version 3.2.0-8
ros-orocos-kdl Version: 1.3.0-0trusty-20151111-0032-+0000

For context, I'm trying to run position only IK on a 4 link finger. I have succesfully run FK as well as a full IK on the chains I create for the fingers.

I'm setting the task space weights as:

task_space_ik_weights = np.diag([1.0, 1.0, 1.0, 0.0, 0.0, 0.0]).tolist()

But I got the same error when giving an identity matrix.

I'm happy to report any more info that you need to help solve this. Also, happy to post this to the forum if my account gets activated.

smits, 2017-02-20:
The assert indicates that the algorithm failed to get a block from a matrix.  Could you post a small example with which we could reproduce the issue, or post a longer backtrace so we can see from which lines the errors originate.

MatthijsBurgh, 2020-05-04:
outdated

closed 2020-05-04 by MatthijsBurgh