orocos/orocos_kinematics_dynamics #89: Eigen all the way down

facontidavide, 2016-11-22:
Hi,

I wonder why KDL::Vector and KDL::Rotation are not just Eigen Vector3d and Matrix3d.

I guess there is a problem of API/ABI compatibility (where the former is easier to solve).
I have the gut feeling that we might get some performance improvement if we drop our own Vector/Rotation/Frame implementation, but I guess that maybe I am missing something...

Regards

Davide

facontidavide, 2016-11-23:
UPDATE:

I implemented the changes and I got a __9% speed improvement__ in the execution speed of chainiksolverpos_lma_demo.

The "trick" is to use Eigen::Vector4d and Eigen::Matrix4d, even is we need 3x1 and 3x3.
This allow Eigen to more effectively vectorize operations and multiplications.

Is there any interest in merging these changes?
Would ABI compatibility be a problem?

smits, 2016-11-23:
The main reason is that KDL::Matrix and KDL::Vector are a lot older then Eigen, we never got to evaluate whether building KDL completely on top of Eigen would be beneficial.  ABI compatibility is definitely a problem, we can only do this for a new minor release, if we break API compatibility we even have to increase our major number.  In this case I think a design doc that outlines these changes might be very beneficial.

facontidavide, 2016-11-23:
Will a working PR be a sufficiently clear "documentation" of the changes?

I need to work a little more on my proof of concept to keep the APi compatible.

The main headache is that KDL::Rotation is row-major, while default Eigen::Matrix is column-major.
This might be a problem for those user that access KDL::Rotation::data directly :(

eaertbel, 2016-11-23:
What about allignment issues for fixed size matrices in Eigen ? See https://eigen.tuxfamily.org/dox/group__DenseMatrixManipulation__Alignement.html

Doesn't this break a lot of user code?

Erwin.

facontidavide, 2016-11-23:
I have applied this solution

https://eigen.tuxfamily.org/dox/group__TopicStructHavingEigenMembers.html

I believe it should solve the issue.

eaertbel, 2016-11-23:
The thing I wanted to emphasize is that all our users should now have this macro EIGEN_MAKE_ALIGNED_OPERATOR_NEW

in their classes.  This makes your suggestion an API change, since most of the user code will need to be adapted, even when

none of the function/method class change.

There are many users that use KDL not only for the forward or backward kinematic routines ( in such case this issue is

not a problem), but also for the basic frame computations inside their own components and software.

At least we should be aware of this issue, and communicate it very well to our users should we make such a change.

best regards,

Erwin.

https://eigen.tuxfamily.org/dox/group__TopicStructHavingEigenMembers.html

facontidavide, 2016-11-24:
I understood what you mean, but it I believe that it is not the case.
User will not need to use EIGEN_MAKE_ALIGNED_OPERATOR_NEW.

I am 90% confident and I am looking forward for a test code that proves me wrong.

On the other hand, there is a much worse problem here than this.
To be sure that the access to single elements doesn't change for KDL::Rotation::data[index], we are obliged to use Eigen::Matrix<double, 3,3, Eigen::RowMajor>.

Unfortunately, this kills the performance gain I measured :(

After all, I am afraid I should just give up on this one.

closed 2016-11-24 by facontidavide