orocos/orocos_kinematics_dynamics #95: catkin dependency

Karsten1987, 2017-03-03:
I was wondering why this package is depending on catkin.

meyerj, 2017-03-05:
I think you are right. The `orocos_kdl` and `python_orocos_kdl` packages do not depend on catkin and the `<run_depend>` line in their package.xml could be removed.

The declared dependency on catkin exists since the first commit that added the `package.xml` files for bloom: https://github.com/orocos/orocos_kinematics_dynamics/commit/f8c31253b2623e22c65abcaf7a887decaac1dd33. I think bloom automatically adds a binary package dependency to `ros-xxx-catkin` for all released packages and it does not need to be declared explicitly unless KDL would rely on catkin for configuration.

The meta-package `orocos_kinematics_dynamics` correctly declares a `<buildtool_depend>` on catkin, but this package is not needed for pure cmake builds.

@smits?

meyerj, 2017-05-31:
I was wrong. The `<run_depend>catkin</run_depend>` line is needed for ROS releases of 3rd-party packages according to REP 136 [1,2,3]. I think the only reason is to have `/opt/ros/kinetic/setup.sh` and other setup files provided by `ros-kinetic-catkin` installed, even if the 3rd-party package itself does not use or invoke catkin.

However, for `orocos_kdl` and `python_orocos_kdl` it might be an option to switch from bloom's `rosdebian` to the `debian` generator, which installs to `/usr` instead [4]. Not sure if this would be properly supported by the ROS build farm and rosdep.

References:
[1] http://www.ros.org/reps/rep-0136.html
[2] http://wiki.ros.org/bloom/Tutorials/ReleaseThirdParty
[3] https://github.com/ros-infrastructure/bloom/issues/87
[4] https://github.com/ros-infrastructure/bloom/issues/382

closed 2017-05-31 by meyerj