orocos/orocos_kinematics_dynamics #96: ImportError: dynamic module does not define module export function (PyInit_PyKDL)

cimenx, 2017-04-15:
Using python 3.5
When I try to import PyKDL `python3 -c "import PyKDL"`, I got that error

dignakov, 2017-12-10:
I'm having the same issue. Built from source in catkin workspace.

foolyc, 2017-12-15:
I'm having the same problem build from SIP.

smits, 2017-12-15:
Hmm, I just checked the SIP documentation, nothing mentions a specific change between python3.(<5) and python3.5 that could explain this issue.  Which SIP version did you use? Maybe we need to provide the init<Module> function ourselves?

foolyc, 2017-12-23:
I  wrapper the KDL with boost.Python in python3.5, it works well.
[https://github.com/foolyc/PyKDL](url)

MatthijsBurgh, 2020-05-04:
This error is a result from opening the python2 bindings in python3. After merging #229, python3 will be supported. Please make sure you are generating the bindings for the correct python version if building from source.

closed 2020-05-04 by MatthijsBurgh