orocos/orocos_kinematics_dynamics #101: liborocos-kdl.so.1.4: cannot open shared object file: No such file or directory

florian02100, 2017-07-06:
Hi,

I'm Florian i work with KDL to solve ik problems.

But when i run my program i've this error : liborocos-kdl.so.1.4: cannot open shared object file: No such file or directory

Nevertheless i've this file in usr/local/lib and LD_LIBRARY = usr/local/lib.
Moreover in my program i've LIBS += "usr/local/lib/*.so".

So i don't understand, if you have any idea ..... ?

Thanks

smits, 2017-10-09:
I think your path is missing a leading slash: "/usr/local/lib"

smits, 2018-03-28:
Closing as this does not seem to be a KDL related problem.

closed 2018-03-28 by smits

prashanthitata, 2021-03-23:
how do we upgrade from 1.3 to 1.4 that the path in /usr/local/lib is not missed?

MatthijsBurgh, 2021-03-23:
@prashanthitata Your question is very unclear and probably not related to this closed issue. So please open a new issue.

P.S. responding to a closed issue, especially when it has been closed for a long time, is a bad habit, unless you are 100% sure it is the same problem.