orocos/orocos_kinematics_dynamics #102: chainjnttojacdotsolver.hpp not available in Indigo and Lunar versions

GuiRitter, 2017-07-12:
Hi everyone.

This file is mentioned in the docs, but isn't present in the Indigo and Lunar installations of `orocos_kdl`.

[Here](http://answers.ros.org/question/266078/where-can-i-get-chainjnttojacdotsolver/) suggested that I added an issue here.

Thanks in advance.

adamconkey, 2017-10-15:
+1 I am similarly trying to use this but it is not in the Kinetic release version. I'm going to try building from source until it's included.

smits, 2017-10-18:
In the form it is in master it cannot be released to Kinetic as it is not ABI compatible, I'll need to find a patchset that includes that solver without breaking ABI backwards compatibility.

smits, 2018-03-28:
This has been released in melodic.

closed 2018-03-28 by smits