orocos/orocos_kinematics_dynamics #103: Dynamic reconfiguring of a Chain object inside a solver e.g., ChainDynParam

xwavex, 2017-07-26:
I need to dynamically change the dynamic parameters i.e. mass of a chain segment, which is used with a solver, such as e.g., ChainDynParam. Since I cannot allocate memory on the heap, it is not feasible for me to always create an new ChainDynParam object with the reconfigured chain as parameter.

What would be the right way to do this?

ahoarau, 2017-07-26:
I think that was the point of #63 , but no news since 2015. I plus one this feature request !

xwavex, 2017-07-26:
I just tried out the branch feature/solvers_use_reference_of_chain referenced by #63. At first glance it seems like that it properly compiles except for the **python_orocos_kdl** part. For me this is totally fine, since I don't use python at all.
In the next days I will integrate this KDL version into my controller code and see what happens.

smits, 2018-03-28:
@xwavex Can you confirm if the latest master (which includes `solvers_use_reference_of_chain`) works for you and close the issue if yes?

MatthijsBurgh, 2020-05-04:
outdated

closed 2020-05-04 by MatthijsBurgh