orocos/orocos_kinematics_dynamics #115: does KDL have python3 binding?

foolyc, 2017-11-14:
In python3, how to use KDL? or any other recommended tools?

ahcorde, 2017-12-15:
If you compile python_orocos_kdl with `catkin` using these flags: `-DPYTHON_VERSION=3.6 -DPYTHON_EXECUTABLE=/usr/bin/python3.5` you should be able to run it with Python3.

foolyc, 2017-12-23:
@ahcorde I complie it with py3.5, but when import the module PyKDL, segmentfault will appear.

Then I wrapper it with boost.Python [https://github.com/foolyc/PyKDL](url)

jacknlliu, 2018-01-24:
I think it needs a standalone release package for python3.

zchen24, 2018-02-07:
PyKDL is compatible with Python 3. Tested with Anaconda Python 3.6 version. 

[image]

To build: 
[image]

smits, 2018-03-28:
@foolyc I cannot reproduce your problem, building the python bindings as @ahcorde suggests works for me, also when using python 3.5.  For the others I wonder how they got everything working without 74c3b80fe6ae

zchen24, 2018-05-13:
@MichaelLutter Can you try the master branch again? Make sure you have a clean build directory for python_orocos_kdl. (e.g. remove your existing build folder and create a new one)

MichaelLutter, 2018-05-13:
@zchen24  Thanks for the fast answer. When I try to compile *6fe130a*, I get compilation errors. I copied the compilation errors below.

Regarding the *Master Branch*, this branched has not changed recently. However, I deleted the build folder and built PyKDL from scratch and it still fails with 'updateInternalDataStructures'. 

"The compiled version is for Python 2, which does not have this PyInit_PyKDL function." What do you mean by that? Can one make PyKDL work under Python3? 

 *6fe130a Compilation Errors:* 
/.../orocos_kinematics_dynamics/python_orocos_kdl/build/PyKDL/sipPyKDLpart2.cpp: In function ‘PyObject* meth_ChainIkSolverVel_pinv_nso_getWeights(PyObject*, PyObject*)’:
/.../orocos_kinematics_dynamics/python_orocos_kdl/build/PyKDL/sipPyKDLpart2.cpp:283:46: error: ‘const class KDL::ChainIkSolverVel_pinv_nso’ has no member named ‘getWeights’; did you mean ‘setWeights’?
             sipRes = new  ::JntArray(sipCpp->getWeights());
                                              ^~~~~~~~~~
                                              setWeights
/.../orocos_kinematics_dynamics/python_orocos_kdl/build/PyKDL/sipPyKDLpart2.cpp: In function ‘PyObject* meth_ChainIkSolverVel_pinv_nso_getOptPos(PyObject*, PyObject*)’:
/.../orocos_kinematics_dynamics/python_orocos_kdl/build/PyKDL/sipPyKDLpart2.cpp:310:46: error: ‘const class KDL::ChainIkSolverVel_pinv_nso’ has no member named ‘getOptPos’; did you mean ‘setOptPos’?
             sipRes = new  ::JntArray(sipCpp->getOptPos());
                                              ^~~~~~~~~
                                              setOptPos
/..../orocos_kinematics_dynamics/python_orocos_kdl/build/PyKDL/sipPyKDLpart2.cpp: In function ‘PyObject* meth_ChainIkSolverVel_pinv_nso_getAlpha(PyObject*, PyObject*)’:
/..../orocos_kinematics_dynamics/python_orocos_kdl/build/PyKDL/sipPyKDLpart2.cpp:337:30: error: ‘const class KDL::ChainIkSolverVel_pinv_nso’ has no member named ‘getAlpha’; did you mean ‘setAlpha’?
             sipRes = sipCpp->getAlpha();

MichaelLutter, 2018-05-13:
Sorry, that was my bad, I compiled the PyKDL with the Version shipped in ROS Lunar. When I adapted the KDL location, I can compile 6fe130a without any errors. However, if I copy the resulting *.so to the python packages, I still get the Python Error:

"ImportError: dynamic module does not define module export function (PyInit_PyKDL)"

Is there any way to solve this error?

[image]

zchen24, 2018-05-13:
This error happens when python 3 tries to load a python 2 module. Try cd to
your python_orocos_kdl build directory (where the PyKDL.so file is), then
run python and try to import PyKDL

MichaelLutter, 2018-05-13:
So I figured out, why python 3 wanted to import the python 2 module. I needed to unset my PYTHONPATH as PYTHONPATH did still point to python 2.

However, if I now import the correct PyKDL.so, I get a new error: 

ImportError: /.../orocos_kinematics_dynamics/python_orocos_kdl/build/PyKDL.so: undefined symbol: _ZNK3KDL20ChainIkSolverPos_LMA8strErrorEi

zchen24, 2018-05-13:
Pretty sure this means that PyKDL.so is linking to the wrong orocos_kdl.so
file. Check your library path.

MichaelLutter, 2018-05-14:
@zchen24 Thanks for the awesome help. It was a problem with the Library Path. 

However, I noticed that PyKDL causes a segmentation fault, if you input a string to any PyKDL function. E.g. kdl.Tree('HelloWorld') causes a segmentation fault for me. I guess it has to do with the change from byte strings to unicode strings in Python 3. 

Did you ever get PyKDL to completely work under Python3?

zchen24, 2018-05-14:
@MichaelLutter 

Did you ever get PyKDL to completely work under Python3? 
Day in and day out. 

I believe you just found a bug,
[image]

smits, 2018-05-15:
@foolyc I assume this issue can be closed?

closed 2018-05-18 by foolyc

asimonov-hitachi, 2019-06-10:
i just tried to follow instructions of building python3 version.
Had to install the SIP that I downloaded from their website (version 4.19.17)
After building/installing I get the following error trying to use PyKDL in python:

```
>>> import PyKDL
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
RuntimeError: the sip module implements API v12.0 to v12.4 but the PyKDL module requires API v12.6
>>> 
```

asimonov-hitachi, 2019-06-10:
actually, realised there was pip-installed sip of wrong version. removed it and the problem is gone

ritalaezza, 2020-05-21:
> I suspect that you did not compile or install orocos_kdl properly. Please remove your orocos_kinematics_dynamics folder altogether. Do a clean checkout of the master branch. And follow the following steps line by line.
> [image]

Hi,

I'm sorry to reopen this thread, but I'm  having troubles understanding some of the intermediate steps to get PyKDL to work on Python3. Could you post some extra screenshots of the cmake options? Also, could you clarify if this works independently from catkin/ROS? I ask this because in the cmake menu I'm seeing my catkin directory as well as some catkin related options. 

Finally, could you clarify the difference between SIP and pybind11, as well as which one I should select and the recommended installation method/version.

MatthijsBurgh, 2020-05-22:
You could check the travis config file. As both python 2 as python 3  versions are build and tested in travis.
It is not depending on catkin.

There should be no difference in available API between PyBind11 and sip. To be honest I have no idea which one has better performance. I recommend the PyBind11 version. As some sip version have a bug which caused #41. By using PyBind11 you are sure you don't have this bug.

ritalaezza, 2020-05-22:
@MatthijsBurgh Great! Thank you, worked like a charm (when I run this in a clean environment, without all the variables created by the ROS setup file which I had sourced in my .zshrc).

Murtaza786-oss, 2020-05-25:
Hello I am trying to read a urdf file but I get error as

`python3: /build/orocos-kdl-mPkyII/orocos-kdl-1.4.0/python_orocos_kdl/PyKDL/std_string.sip:52: int convertTo_std_string(PyObject*, void**, int*, PyObject*): Assertion `PyUnicode_Check(s)' failed.
`
I am using the recommended Ubuntu 20.04 (Focal Fossa) with ROS Noetic

It never used to happen on melodic with python2.7. I am not sure whether its a problem with the package urdf_parser_py or is it with PyKDL?. 

I  get the error when i try to execute,

    ```
from __future__ import print_function
    import PyKDL as kdl
    import urdf_parser_py.urdf as urdf
    
         with open(filename) as urdfFile:
                    return treeFromUrdfModel(urdf.URDF.from_xml_string(urdfFile.read()))

```

I think my problem is similar to this, https://github.com/orocos/orocos_kinematics_dynamics/issues/115#issuecomment-388792903

and I think this is the solution, https://github.com/orocos/orocos_kinematics_dynamics/pull/145#issue-187920168

But I am not sure how should I edit and incorporate the changes mentioned above.

My urdf file is,

    ```
<?xml version="1.0"?>
    <!-- =================================================================================== -->
    <!-- |    This document was autogenerated by xacro from ur5_joint_limited_robot.urdf.xacro | -->
    <!-- |    EDITING THIS FILE BY HAND IS NOT RECOMMENDED                                 | -->
    <!-- =================================================================================== -->
    <robot name="ur5">
      <gazebo>
        <plugin filename="libgazebo_ros_control.so" name="ros_control">
          <!-- <robotNamespace>/</robotNamespace> -->
          <robotSimType>gazebo_ros_control/DefaultRobotHWSim</robotSimType>
          <legacyModeNS>true</legacyModeNS>
        </plugin>
        <!--
        <plugin name="gazebo_ros_power_monitor_controller" filename="libgazebo_ros_power_monitor.so">
          <alwaysOn>true</alwaysOn>
          <updateRate>1.0</updateRate>
          <timeout>5</timeout>
          <powerStateTopic>power_state</powerStateTopic>
          <powerStateRate>10.0</powerStateRate>
          <fullChargeCapacity>87.78</fullChargeCapacity>     
          <dischargeRate>-474</dischargeRate>
          <chargeRate>525</chargeRate>
          <dischargeVoltage>15.52</dischargeVoltage>
          <chargeVoltage>16.41</chargeVoltage>
        </plugin>
```

MatthijsBurgh, 2020-05-25:
@Murtaza786-oss Please open a new issue. It is completely not related to the original issue.