orocos/orocos_kinematics_dynamics #129: PyKDL with Ubuntu 16.04 does not access correct memory when creating or copying frames.

daenny, 2018-02-08:
When copying or creating a new KDL frame and directly accessing the vector or rotation matrix, the result is not correct. 
It seems as if not the correct memory is accessed. When splitting the creation and the access in two lines it works as expected. 
See code below (tested with: `Linux 4.13.0-32-generic #35~16.04.1-Ubuntu`)
```python
# coding: utf-8
import PyKDL as kdl
a = kdl.Frame(kdl.Rotation.RPY(1,2,3), kdl.Vector(1,2,3)).p
a
# [6.91476e-310,6.91476e-310,6.91476e-310]
a = kdl.Frame(kdl.Rotation.RPY(1,2,3), kdl.Vector(1,2,3))
a.p
# [           1,           2,           3]
b = (kdl.Frame() * a).p
b
# [6.91476e-310,6.91476e-310,6.91476e-310]
import copy
copy.deepcopy(a).p
# [6.91476e-310,6.91476e-310,6.91476e-310]
```
Under 14.04 this code works as expected.

daenny, 2018-02-22:
It seems that this is as SIP issue. See the other issue. SIP above 4.16.8 generates python bindings which do not work correctly and above 4.19 it even segfaults on import...

smits, 2018-03-28:
I can reproduce this, no idea on how to fix this yet though.

mrana6, 2019-06-17:
Has anyone fixed this yet? Im getting seg faults on running PyKDL on 16.04. Im using sip 4.19.8

MatthijsBurgh, 2020-01-16:
Please check [my comment](https://github.com/orocos/orocos_kinematics_dynamics/issues/41#issuecomment-573624235) in #41 for possible fixes

MatthijsBurgh, 2020-03-29:
PR for PyKDL based on PyBind11 is open: #229

`python-sip` fix is also pending for approval in ubuntu launchpad, so that should also be released in a few weeks.

MatthijsBurgh, 2020-04-01:
A fixed version of `python-sip` is now available on `updates` both for Xenial and Bionic.

closed 2021-02-04 by daenny