orocos/orocos_kinematics_dynamics #136: PyKDL not building from source

MatthijsBurgh, 2018-03-27:
I have tried SIP 4.15.5, 4.16.8 and 4.19.8.
```
Errors     << python_orocos_kdl:make /home/amigo/ros/kinetic/system/logs/python_orocos_kdl/build.make.003.log     
/home/amigo/ros/kinetic/system/build/python_orocos_kdl/PyKDL/sipPyKDLpart1.cpp: In member function ‘void sipChainJntToJacSolver::updateInternalDataStructures()’:
/home/amigo/ros/kinetic/system/build/python_orocos_kdl/PyKDL/sipPyKDLpart1.cpp:2023:9: error: ‘updateInternalDataStructures’ is not a member of ‘KDL::ChainJntToJacSolver’
/home/amigo/ros/kinetic/system/build/python_orocos_kdl/PyKDL/sipPyKDLpart1.cpp: In function ‘PyObject* meth_ChainJntToJacSolver_updateInternalDataStructures(PyObject*, PyObject*)’:
/home/amigo/ros/kinetic/system/build/python_orocos_kdl/PyKDL/sipPyKDLpart1.cpp:2080:59: error: ‘class KDL::ChainJntToJacSolver’ has no member named ‘updateInternalDataStructures’
/home/amigo/ros/kinetic/system/build/python_orocos_kdl/PyKDL/sipPyKDLpart1.cpp:2080:100: error: ‘class KDL::ChainJntToJacSolver’ has no member named ‘updateInternalDataStructures’
/home/amigo/ros/kinetic/system/build/python_orocos_kdl/PyKDL/sipPyKDLpart1.cpp: In member function ‘void sipChainIkSolverVel_pinv_givens::updateInternalDataStructures()’:
/home/amigo/ros/kinetic/system/build/python_orocos_kdl/PyKDL/sipPyKDLpart1.cpp:2305:9: error: ‘updateInternalDataStructures’ is not a member of ‘KDL::ChainIkSolverVel_pinv_givens’
     extern void sipVH_PyKDL_0(sip_gilstate_t, sipVirtErrorHandlerFunc, sipSimpleWrapper *, PyObject *);
         ^
/home/amigo/ros/kinetic/system/build/python_orocos_kdl/PyKDL/sipPyKDLpart1.cpp: In function ‘PyObject* meth_ChainIkSolverVel_pinv_givens_updateInternalDataStructures(PyObject*, PyObject*)’:
/home/amigo/ros/kinetic/system/build/python_orocos_kdl/PyKDL/sipPyKDLpart1.cpp:2380:68: error: ‘class KDL::ChainIkSolverVel_pinv_givens’ has no member named ‘updateInternalDataStructures’
/home/amigo/ros/kinetic/system/build/python_orocos_kdl/PyKDL/sipPyKDLpart1.cpp:2380:109: error: ‘class KDL::ChainIkSolverVel_pinv_givens’ has no member named ‘updateInternalDataStructures’
make[2]: *** [CMakeFiles/python_module_PyKDL.dir/PyKDL/sipPyKDLpart1.o] Error 1
make[2]: *** Waiting for unfinished jobs....
/home/amigo/ros/kinetic/system/build/python_orocos_kdl/PyKDL/sipPyKDLpart3.cpp: In member function ‘void sipChainIkSolverPos_NR::updateInternalDataStructures()’:
/home/amigo/ros/kinetic/system/build/python_orocos_kdl/PyKDL/sipPyKDLpart3.cpp:87:9: error: ‘updateInternalDataStructures’ is not a member of ‘KDL::ChainIkSolverPos_NR’
         ChainIkSolverPos_NR::updateInternalDataStructures();
         ^
/home/amigo/ros/kinetic/system/build/python_orocos_kdl/PyKDL/sipPyKDLpart3.cpp: In function ‘PyObject* meth_ChainIkSolverPos_NR_updateInternalDataStructures(PyObject*, PyObject*)’:
/home/amigo/ros/kinetic/system/build/python_orocos_kdl/PyKDL/sipPyKDLpart3.cpp:162:59: error: ‘class KDL::ChainIkSolverPos_NR’ has no member named ‘updateInternalDataStructures’
             (sipSelfWasArg ? sipCpp->ChainIkSolverPos_NR::updateInternalDataStructures() : sipCpp->updateInternalDataStructures());
                                                           ^
/home/amigo/ros/kinetic/system/build/python_orocos_kdl/PyKDL/sipPyKDLpart3.cpp:162:100: error: ‘class KDL::ChainIkSolverPos_NR’ has no member named ‘updateInternalDataStructures’
             (sipSelfWasArg ? sipCpp->ChainIkSolverPos_NR::updateInternalDataStructures() : sipCpp->updateInternalDataStructures());
                                                                                                    ^
/home/amigo/ros/kinetic/system/build/python_orocos_kdl/PyKDL/sipPyKDLpart3.cpp: In function ‘PyObject* meth_ChainIkSolverVel_updateInternalDataStructures(PyObject*, PyObject*)’:
/home/amigo/ros/kinetic/system/build/python_orocos_kdl/PyKDL/sipPyKDLpart3.cpp:528:21: error: ‘class KDL::ChainIkSolverVel’ has no member named ‘updateInternalDataStructures’
/home/amigo/ros/kinetic/system/build/python_orocos_kdl/PyKDL/sipPyKDLpart3.cpp: In function ‘PyObject* meth_ChainIkSolverPos_updateInternalDataStructures(PyObject*, PyObject*)’:
/home/amigo/ros/kinetic/system/build/python_orocos_kdl/PyKDL/sipPyKDLpart3.cpp:827:21: error: ‘class KDL::ChainIkSolverPos’ has no member named ‘updateInternalDataStructures’
             Py_INCREF(Py_None);
                     ^
/home/amigo/ros/kinetic/system/build/python_orocos_kdl/PyKDL/sipPyKDLpart3.cpp: In member function ‘void sipChainFkSolverVel_recursive::updateInternalDataStructures()’:
/home/amigo/ros/kinetic/system/build/python_orocos_kdl/PyKDL/sipPyKDLpart3.cpp:1044:9: error: ‘updateInternalDataStructures’ is not a member of ‘KDL::ChainFkSolverVel_recursive’
/home/amigo/ros/kinetic/system/build/python_orocos_kdl/PyKDL/sipPyKDLpart3.cpp: In function ‘PyObject* meth_ChainFkSolverVel_recursive_updateInternalDataStructures(PyObject*, PyObject*)’:
/home/amigo/ros/kinetic/system/build/python_orocos_kdl/PyKDL/sipPyKDLpart3.cpp:1119:66: error: ‘class KDL::ChainFkSolverVel_recursive’ has no member named ‘updateInternalDataStructures’
/home/amigo/ros/kinetic/system/build/python_orocos_kdl/PyKDL/sipPyKDLpart3.cpp:1119:107: error: ‘class KDL::ChainFkSolverVel_recursive’ has no member named ‘updateInternalDataStructures’
/home/amigo/ros/kinetic/system/build/python_orocos_kdl/PyKDL/sipPyKDLpart3.cpp: In member function ‘void sipChainFkSolverPos_recursive::updateInternalDataStructures()’:
/home/amigo/ros/kinetic/system/build/python_orocos_kdl/PyKDL/sipPyKDLpart3.cpp:1344:9: error: ‘updateInternalDataStructures’ is not a member of ‘KDL::ChainFkSolverPos_recursive’
     extern void sipVH_PyKDL_0(sip_gilstate_t, sipVirtErrorHandlerFunc, sipSimpleWrapper *, PyObject *);
         ^
/home/amigo/ros/kinetic/system/build/python_orocos_kdl/PyKDL/sipPyKDLpart3.cpp: In function ‘PyObject* meth_ChainFkSolverPos_recursive_updateInternalDataStructures(PyObject*, PyObject*)’:
/home/amigo/ros/kinetic/system/build/python_orocos_kdl/PyKDL/sipPyKDLpart3.cpp:1419:66: error: ‘class KDL::ChainFkSolverPos_recursive’ has no member named ‘updateInternalDataStructures’
/home/amigo/ros/kinetic/system/build/python_orocos_kdl/PyKDL/sipPyKDLpart3.cpp:1419:107: error: ‘class KDL::ChainFkSolverPos_recursive’ has no member named ‘updateInternalDataStructures’
make[2]: *** [CMakeFiles/python_module_PyKDL.dir/PyKDL/sipPyKDLpart3.o] Error 1
/home/amigo/ros/kinetic/system/build/python_orocos_kdl/PyKDL/sipPyKDLpart2.cpp: In member function ‘void sipChainIkSolverVel_pinv_nso::updateInternalDataStructures()’:
/home/amigo/ros/kinetic/system/build/python_orocos_kdl/PyKDL/sipPyKDLpart2.cpp:127:9: error: ‘updateInternalDataStructures’ is not a member of ‘KDL::ChainIkSolverVel_pinv_nso’
         ChainIkSolverVel_pinv_nso::updateInternalDataStructures();
         ^
/home/amigo/ros/kinetic/system/build/python_orocos_kdl/PyKDL/sipPyKDLpart2.cpp: In function ‘PyObject* meth_ChainIkSolverVel_pinv_nso_updateInternalDataStructures(PyObject*, PyObject*)’:
/home/amigo/ros/kinetic/system/build/python_orocos_kdl/PyKDL/sipPyKDLpart2.cpp:202:65: error: ‘class KDL::ChainIkSolverVel_pinv_nso’ has no member named ‘updateInternalDataStructures’
             (sipSelfWasArg ? sipCpp->ChainIkSolverVel_pinv_nso::updateInternalDataStructures() : sipCpp->updateInternalDataStructures());
                                                                 ^
/home/amigo/ros/kinetic/system/build/python_orocos_kdl/PyKDL/sipPyKDLpart2.cpp:202:106: error: ‘class KDL::ChainIkSolverVel_pinv_nso’ has no member named ‘updateInternalDataStructures’
             (sipSelfWasArg ? sipCpp->ChainIkSolverVel_pinv_nso::updateInternalDataStructures() : sipCpp->updateInternalDataStructures());
                                                                                                          ^
/home/amigo/ros/kinetic/system/build/python_orocos_kdl/PyKDL/sipPyKDLpart2.cpp: In member function ‘void sipChainIkSolverPos_LMA::updateInternalDataStructures()’:
/home/amigo/ros/kinetic/system/build/python_orocos_kdl/PyKDL/sipPyKDLpart2.cpp:619:9: error: ‘updateInternalDataStructures’ is not a member of ‘KDL::ChainIkSolverPos_LMA’
         return;
         ^
/home/amigo/ros/kinetic/system/build/python_orocos_kdl/PyKDL/sipPyKDLpart2.cpp: In function ‘PyObject* meth_ChainIkSolverPos_LMA_updateInternalDataStructures(PyObject*, PyObject*)’:
/home/amigo/ros/kinetic/system/build/python_orocos_kdl/PyKDL/sipPyKDLpart2.cpp:694:60: error: ‘class KDL::ChainIkSolverPos_LMA’ has no member named ‘updateInternalDataStructures’
[... cut ...]
/home/amigo/ros/kinetic/system/build/python_orocos_kdl/PyKDL/sipPyKDLpart2.cpp:1778:103: error: ‘class KDL::ChainIkSolverPos_NR_JL’ has no member named ‘updateInternalDataStructures’
make[2]: *** [CMakeFiles/python_module_PyKDL.dir/PyKDL/sipPyKDLpart2.o] Error 1
make[1]: *** [CMakeFiles/python_module_PyKDL.dir/all] Error 2
make: *** [all] Error 2
cd /home/amigo/ros/kinetic/system/build/python_orocos_kdl; catkin build --get-env python_orocos_kdl | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
```

MatthijsBurgh, 2018-03-27:
Sorry, It was solved by updating libcppunit-dev to the newest version. Although #129 is still an issue.

closed 2018-03-27 by MatthijsBurgh

ekorudiawan, 2020-02-26:
> Sorry, It was solved by updating libcppunit-dev to the newest version. Although #129 is still an issue.

Which version of libcppunit-dev did you use? I also got a problem when trying to compile PyKDL with python 3 support on ROS kinetic

MatthijsBurgh, 2020-02-26:
This is almost 2 years ago. I don't know anymore.