orocos/orocos_kinematics_dynamics #147: Should ChainIkSolverPos__LMA store a KDL::Chain as a const ref?

diogoalmeida, 2018-05-27:
When the Levenberg-Marquardt solver stores the kinematic chain as a const ref,

https://github.com/orocos/orocos_kinematics_dynamics/blob/a7a8d282a5a94948176f2c5af58fbd047d849030/orocos_kdl/src/chainiksolverpos_lma.hpp#L160

It allows scenarios where the chain passed in the constructor is destroyed, and the solver will eventually segfault when its called, without warning for what has happened. If the intention is to not copy the chain, a shared pointer should be used. 

However, the other solvers seem to be storing a copy of the chain, and don't have this issue. 

Just spent a couple hours trying to understand why suddenly my IK solver was segfaulting :)

smits, 2018-12-27:
Sorry to here about time being wasted over this.  Having a reference instead of a copy is intentional see https://github.com/orocos/orocos_kinematics_dynamics/pull/63 .  A solution with shared pointers is welcomed, but I imagine this will most probably break the API

diogoalmeida, 2018-12-27:
Just out of curiosity, what is the reasoning behind the const refs?

Thanks for taking the time to reply, happy holidays!

smits, 2018-12-27:
It was mostly to support changing masses and end effectors including extra
joints without having to recreate the solvers.

disRecord, 2019-01-20:
From my viewpoint it is bad design decision. I commented it here https://github.com/orocos/orocos_kinematics_dynamics/pull/63#issuecomment-455871763