orocos/orocos_kinematics_dynamics #148: jacobiandottest fails on i686

morxa, 2018-05-30:
The jacobiandottest fails with the following error ([full log](https://kojipkgs.fedoraproject.org//work/tasks/385/27300385/build.log)):

```
5/7 Test #5: jacobiantest .....................   Passed    0.00 sec
test 6
    Start 6: jacobiandottest
6: Test command: /builddir/build/BUILD/orocos_kinematics_dynamics-1.4.0/orocos_kdl/tests/jacobiandottest
6: Test timeout computed to be: 10000000
6: ...... dt:0.001 err:0.00309775 eps_diff_vs_solver:0.003
6:  dt:0.01 err:0.00842029 eps_diff_vs_solver:0.03
6:  dt:0.1 err:0.0108403 eps_diff_vs_solver:0.3
6: F....
6: 
6/7 Test #6: jacobiandottest ..................***Failed    0.06 sec
```

It looks like the error is just slightly too high (`0.00309775` vs `0.003`). Looking at the code, the threshold seemd to be picked arbitrarily:

```
double eps_diff_vs_solver = 3.0*dt; // Apparently :)
```

Maybe the threshold just needs to be increased? Or is this an actual bug?

The test succeeds on all other architectures:
https://koji.fedoraproject.org/koji/taskinfo?taskID=27300380

ahoarau, 2018-05-30:
That was an empirical value I found, 100% for changing to 4.0 if it works everywhere.

closed 2018-06-29 by smits

ahoarau, 2018-06-29:
I'm still very curious on why this would fail on a different arch. Any thoughts @morxa ?

morxa, 2018-06-29:
I can only guess it's related to different floating point precision on different architectures, e.g., because of different sizes of the base types such as `sizeof(float)`. Not sure what the exact difference is here, though.

PR #149 (2018-06-29): Increase threshold tolerance in jacobiandottest
This fixes test failures on i686, where the error would be just slightly
higher than the allowed value.

This fixes #148.