orocos/orocos_kinematics_dynamics #168: PyKDL on Windows

hosseinahm, 2018-11-06:
I've compiled both orocos_kdl and python_orocos_kdl on windows but cannot import PyKDL module into python saying "No module named PyKDL"
I have tried VC++15 and VC++14 compilers and Python2.7 and make sure generated PyKDL.dll file is on the PYTHONPATH and PATH search directories.
Has anybody succeeded in using PyKDL binding in windows?

seanyen-msft, 2019-02-27:
Per my experience, usually the python module on Windows should be ended with `.pyd` extension. You can rename `PyKDL.dll` into `PyKDL.pyd` to verify.

maxsvetlik, 2019-03-04:
@seanyen-msft 's suggestion worked for me. In my case the `PyKDL.dll` file lived at `C:\opt\ros\melodic\x64\lib\site-packages` from an install of `orocos_kdl` from [nuget](https://roswin.azurewebsites.net/packages/orocos_kdl/1.3.1.1809242322)

MatthijsBurgh, 2020-05-04:
Fixed by #189. Please open new issue in case you still experience issues.

closed 2020-05-04 by MatthijsBurgh