orocos/orocos_kinematics_dynamics #170: Rigid body inertia frame

mrslvg, 2018-11-28:
I spent several days (still not succeed) trying to understand in which reference frame should the RigidBodyInertia be defined when creating a new Segment. Has someone ever had the same problem? 
The 'official' way  ([link](http://www.orocos.org/wiki/main-page/kdl-wiki/user-manual/kinematic-trees)) doesn't work for me. I already tried (with no success) the solution suggested by [this page](http://www.orocos.org/forum/orocos/orocos-users/problem-dynamic-model-and-inverse-dynamic-solver-kdl). 
I would like very much to understand this an write a clearer documentation of the Segment class.

francofusco, 2019-01-09:
I agree that the documentation is a little misleading. In the [wiki page](http://www.orocos.org/wiki/main-page/kdl-wiki/user-manual/kinematic-trees) you linked it indeed seems that inertia is referring to the "parent reference frame". However, this does not seem to be the real usage: 
as pointed out by Santiago in the [other link you included](http://www.orocos.org/forum/orocos/orocos-users/problem-dynamic-model-and-inverse-dynamic-solver-kdl), inertial properties are expressed at the tip frame (that is, "after" the joint). Indeed, if a joint "moves" then its effort directly depends on the inertia of the segment who "owns" the joint in the implementation of [`ChainIdSolver_RNE`](https://github.com/orocos/orocos_kinematics_dynamics/blob/master/orocos_kdl/src/chainidsolver_recursive_newton_euler.cpp). I specifically refer to lines [84](https://github.com/orocos/orocos_kinematics_dynamics/blob/9ff3052d6b2799bc039bccdab8f85b61367026eb/orocos_kdl/src/chainidsolver_recursive_newton_euler.cpp#L84), in which inertial forces are added to the wrench applied to the current segment and [91](https://github.com/orocos/orocos_kinematics_dynamics/blob/9ff3052d6b2799bc039bccdab8f85b61367026eb/orocos_kdl/src/chainidsolver_recursive_newton_euler.cpp#L91) which computes the effort for the joint.

Since you said that Santiago's post did not help you, perhaps you can give more details about the problem you are facing and I will try to help you - although I am not really an expert of KDL... but two heads are better than one, isn't it?

In addition I would argue that the way the tip transformation is given to a Segment and its documentation are not consistent. From what I understand from the code, the Frame object passed to the constructor corresponds to the desired transformation between segment's reference and tip **when the joint is in the zero position**. From the doc, it rather seems that the passed tip corresponds to a transformation from the joint to the tip, and that the overall transformation for a segment will be given by the composition of:
1. Joint offset translation
2. Joint transformation (rotation/translation according to its type)
3. Additional tip transformation (which from the doc seems to be the one to be given to the constructor).
From the code, instead, the tip you pass upon construction is not the one used in step 3. I think that this might be another factor you should consider when modelling your tree.

If I receive confirmation that what I said is correct, I will try to update the documentation!

traversaro, 2019-01-09:
Related issue on how the RigidBodyInertia is interpreted  in ROS/URDF when creating a KDL model: https://github.com/ros/robot_model/pull/66 .