orocos/orocos_kinematics_dynamics #190: License in package.xml

shr-project, 2019-08-14:
Currently
https://github.com/orocos/orocos_kinematics_dynamics/blob/master/python_orocos_kdl/package.xml#L10
https://github.com/orocos/orocos_kinematics_dynamics/blob/master/orocos_kinematics_dynamics/package.xml#L11
https://github.com/orocos/orocos_kinematics_dynamics/blob/master/orocos_kdl/package.xml#L10
says only LGPL, while:
https://github.com/orocos/orocos_kinematics_dynamics/blob/master/python_orocos_kdl/PyKDL/PyKDL.sip#L8
https://github.com/orocos/orocos_kinematics_dynamics/blob/master/orocos_kdl/COPYING
https://github.com/orocos/orocos_kinematics_dynamics/blob/master/orocos_kdl/src/articulatedbodyinertia.cpp#L8
says LGPL-2.1+

superflore maps this to LGPL-2 in:
https://github.com/ros-infrastructure/superflore/blob/master/superflore/utils.py#L179
and LGPL-2 isn't valid SPDX Identifier in:
https://spdx.org/licenses/
so it would be better to declare the license correctly at the source - in package.xml.

I've already reported it here https://github.com/ros2-gbp/orocos_kinematics_dynamics-release/issues/2 but was redirected to this repo.

MatthijsBurgh, 2021-06-18:
@smits Which LPGL is correct?

smits, 2021-06-21:
I would say LGPL-2.1+: https://spdx.org/licenses/LGPL-2.1-or-later.html

PR #342 (2021-06-22): Update license tags
Fixes #190
closed 2021-06-22 by MatthijsBurgh