orocos/orocos_kinematics_dynamics #191: Could not find SIP

eugval, 2019-08-27:
Hello, 

I am trying to follow the steps described in [this issue](https://github.com/orocos/orocos_kinematics_dynamics/issues/115) in order to get a PyKDL binding with python3 as the one I installed through ros only imports correctly with python2 .

I cloned the master branch in this repository and followed step by step the procedure described in the thread:
[image]

When building python_orocos_kdl, I get the following error:
[image]

I have installed SIP with pip3 : `pip3 install --user SIP` .
When I also type : `pip install --user SIP`, but I  the following : 
[image]

my `PYTHONPATH` is the following: 
[image]

I am quite confused about this, I suspect I am having a path issue but I cannot figure it out! 

I would greatly appreciate any help understanding what is happening here, and how can I fix this as I am stuck trying to build the package!

eugval, 2019-08-27:
My issue seemed to be that SIP was compiled and installed for python2, not python3. 

This has nothing to do with the pip package that is just a binding (Please correct me if I'm wrong).

To install it for python3, i followed the instructions in [here](https://github.com/Ultimaker/libArcus/issues/59).

After this, the build worked!

MatthijsBurgh, 2020-03-04:
please close this issue

closed 2020-03-04 by meyerj