orocos/orocos_kinematics_dynamics #203: When in the orocos_kinematics_dynamics/python_orocos_kdl/build to execute the command "make"

TFLQW, 2019-11-29:
/home/tiboy/orocos_kinematics_dynamics/python_orocos_kdl/build/PyKDL/sipPyKDLpart1.cpp: In member function ‘void sipChainIdSolver_RNE::updateInternalDataStructures()’:
/home/tiboy/orocos_kinematics_dynamics/python_orocos_kdl/build/PyKDL/sipPyKDLpart1.cpp:2034:9: error: ‘updateInternalDataStructures’ is not a member of ‘KDL::ChainIdSolver_RNE’
/home/tiboy/orocos_kinematics_dynamics/python_orocos_kdl/build/PyKDL/sipPyKDLpart1.cpp: In function ‘PyObject* meth_ChainIdSolver_RNE_updateInternalDataStructures(PyObject*, PyObject*)’:
/home/tiboy/orocos_kinematics_dynamics/python_orocos_kdl/build/PyKDL/sipPyKDLpart1.cpp:2099:57: error: ‘class KDL::ChainIdSolver_RNE’ has no member named ‘updateInternalDataStructures’
/home/tiboy/orocos_kinematics_dynamics/python_orocos_kdl/build/PyKDL/sipPyKDLpart1.cpp:2099:98: error: ‘class KDL::ChainIdSolver_RNE’ has no member named ‘updateInternalDataStructures’
/home/tiboy/orocos_kinematics_dynamics/python_orocos_kdl/build/PyKDL/sipPyKDLpart1.cpp: In function ‘void release_ChainIdSolver_RNE(void*, int)’:
/home/tiboy/orocos_kinematics_dynamics/python_orocos_kdl/build/PyKDL/sipPyKDLpart1.cpp:2136:61: warning: deleting object of polymorphic class type ‘KDL::ChainIdSolver_RNE’ which has non-virtual destructor might cause undefined behaviour [-Wdelete-non-virtual-dtor]
/home/tiboy/orocos_kinematics_dynamics/python_orocos_kdl/build/PyKDL/sipPyKDLpart1.cpp: In function ‘void* cast_ChainIdSolver(void*, const sipTypeDef*)’:
/home/tiboy/orocos_kinematics_dynamics/python_orocos_kdl/build/PyKDL/sipPyKDLpart1.cpp:2285:70: error: ‘SolverI’ was not declared in this scope
/home/tiboy/orocos_kinematics_dynamics/python_orocos_kdl/build/PyKDL/sipPyKDLpart1.cpp:2285:79: error: expected primary-expression before ‘)’ token
/home/tiboy/orocos_kinematics_dynamics/python_orocos_kdl/build/PyKDL/sipPyKDLpart1.cpp:2285:95: error: expected primary-expression before ‘*’ token
/home/tiboy/orocos_kinematics_dynamics/python_orocos_kdl/build/PyKDL/sipPyKDLpart1.cpp:2285:96: error: expected primary-expression before ‘)’ token
/home/tiboy/orocos_kinematics_dynamics/python_orocos_kdl/build/PyKDL/sipPyKDLpart1.cpp: In function ‘void release_ChainIdSolver(void*, int)’:
/home/tiboy/orocos_kinematics_dynamics/python_orocos_kdl/build/PyKDL/sipPyKDLpart1.cpp:2296:53: warning: deleting object of abstract class type ‘KDL::ChainIdSolver’ which has non-virtual destructor will cause undefined behaviour [-Wdelete-non-virtual-dtor]
 extern "C" {static void dealloc_ChainIdSolver(sipSimpleWrapper *);}
                                                     ^
CMakeFiles/python_module_PyKDL.dir/build.make:113: recipe for target 'CMakeFiles/python_module_PyKDL.dir/PyKDL/sipPyKDLpart1.cpp.o' failed
make[2]: *** [CMakeFiles/python_module_PyKDL.dir/PyKDL/sipPyKDLpart1.cpp.o] Error 1
CMakeFiles/Makefile2:163: recipe for target 'CMakeFiles/python_module_PyKDL.dir/all' failed
make[1]: *** [CMakeFiles/python_module_PyKDL.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2

I push this issue, I have followed the guide in the  	.travis.yml

wangjunfeng90214, 2020-02-24:
I met the same issue. My system is Ubuntu 16.04, python3.5, sip 4.17

63445538, 2020-03-17:
Same issue. Ubuntu 16.04, python2.7, sip 4.17

marcos452, 2020-04-12:
> I met the same issue. My system is Ubuntu 16.04, python3.5, sip 4.17

hi guys，did u finish this problem?

MatthijsBurgh, 2020-05-04:
Right now python 3 is not supported. Support will be added after merging #229. Also sip will be replace by PyBind11. Hopefully this will fix your issues. Closing the issue for now.

closed 2020-05-04 by MatthijsBurgh