orocos/orocos_kinematics_dynamics #219: PyBind11 instead of SIP

MatthijsBurgh, 2020-02-08:
Although the bugs in SIP have been fixed https://github.com/orocos/orocos_kinematics_dynamics/issues/41#issuecomment-578111408, a new release of `python-sip` is needed. A bug report if filed on LaunchPad https://github.com/orocos/orocos_kinematics_dynamics/issues/41#issuecomment-581128891, but no progress there yet.

A possible solution would be switching to [PyBind11](https://github.com/pybind/pybind11), as this needs to build from source, this can easily been included with submodules.
@zchen24 did most of the work, which I included in my fork, check https://github.com/MatthijsBurgh/orocos_kinematics_dynamics/tree/pybind11 for the current status.

jf---, 2020-03-24:
This is an exciting avenue, do hope to see the `pybind11` land in master.
The perspective of sidestepping binder specific code, but being able to just use plain C++ is a huge win, impressive work...

MatthijsBurgh, 2020-03-24:
@jf--- Thanks. I plan to finish my work by the end of the week or the start of next week. After which I will open a PR to merge it into master.

MatthijsBurgh, 2020-03-29:
PR is open: #229

PR #229 (2020-05-06): PyKDL: PyBind11 based
This PR includes the following:
- PyKDL based on PyBind11, which is an option, `BUILD_PYKDL_PYBIND11`. So SIP is still there, but PyBind11 is the default. PyBind11 is included as submodule.
- Improvements on PyKDL based on SIP
    - Moved `std::vector` support to separate file.
    - Fixed custom methodcode which contained a return. Which isn't allowed in SIP. Mostly `__getitem__` and `__setitem__` functions.
    - Added some extra constructors, `__repr__` and equality operators.
- Improvements on kdl, which were needed to fix issues in PyKDL.
- Extending the PyKDL tests. [... cut]
closed 2020-05-06 by MatthijsBurgh