orocos/orocos_kinematics_dynamics #220: Error during Compilation of PyKDL

sdsonawani, 2020-02-15:
Hi I am using Ubuntu 18.04 and Trying to install PyKDL for Python3. I am getting following error during compilation. 

~~~
error: ‘const class KDL::Joint’ has no member named ‘getInertia’; did you mean ‘inertia’?
     /* Raise an exception if the arguments couldn't be parsed. */
                              ^~~~~~~~~~
                              inertia
~~~
Any help is appreciated!

meyerj, 2020-02-22:
Which version of KDL do you have installed? Ubuntu 18.04 now provides its own version of [orocos-kdl](https://packages.ubuntu.com/source/bionic/orocos-kdl), but unfortunately it is not up-to-date and also does not include the Python bindings.

You must use the corresponding version in this repo, [1.3.1](https://github.com/orocos/orocos_kinematics_dynamics/releases/tag/v1.3.1). That _might_ work if you only build package `python_orocos_kdl` against the system version of `liborocos-kdl-dev`.

If you do not depend on `liborocos-kdl-dev` I recommend to build and install both, `orocos_kdl` and `python_orocos_kdl` from source from the same version of this repository.

MatthijsBurgh, 2020-03-02:
Please add `python3` to the issue title.

sdsonawani, 2020-03-04:
Thank you @meyerj 
I tackled this issue following way: 
I was able to compile PyKDL by checking out the old commit.
```
- git clone https://github.com/orocos/orocos_kinematics_dynamics.git
- cd orocos_kinematics_dynamics
- git checkout 1ae45bb2e0821586e74047b01f7cb48d913204f4
```
and then compiled the whole thing for python3. This commit is from my old machine where I had already compiled pykdl long time ago. I do not know the available support for this commit. 

Thanks!!

closed 2020-03-05 by meyerj