orocos/orocos_kinematics_dynamics #224: Generate header files for PyKDL.so

gpldecha, 2020-02-27:
In python I would like to pass a PyKDL object as an argument to a boost::python to c++ object.

For instance in python I have the following:

```python
   chain = PyKDL.Chain()
   ForceController(chain)
```

My c++ python object is a follows (pseudo code)

```c++

#include <boost/python.hpp>
#include <boost/python/extract.hpp>
#include <boost/python/suite/indexing/vector_indexing_suite.hpp>
#include <kdl/kdl.hpp>
#include <kdl/jntarray.hpp>

namespace py = boost::python;
using namespace KDL;

BOOST_PYTHON_MODULE(pycontrollers){
        py::numpy::initialize();
        py::class_<fc::ForceControllerModule>("ForceController", py::init<KDL::Chain>());
```
I get the following error:

```bash
Boost.Python.ArgumentError: Python argument types in
    ForceController.__init__(ForceController, Chain)
did not match C++ signature:
    __init__(_object*, KDL::Chain)
```
The issue I think is that PyKDL.so functions have a different signature than the standard KDL library header functions. To be able to use a python PyKDL.so object in my boost python object I would need to have access to the headers for PyKDL.so.

Is there a way to have sig generate these header functions ?

MatthijsBurgh, 2020-03-02:
Why do you need PyKDL in c++ instead of the KDL c++ library?

gpldecha, 2020-03-02:
@MatthijsBurgh Because I am passing a PyKDL object from python to c++ and the PyKDL objects has a different signature than the KDL c++ library. That is why I am getting the "did not match C++ signature". 

To resolve this error (from my understanding), I would need to have access the the PyKDL signature of Chain which is **not** KDL::Chain. So I was wondering if it was possible to generate the header files associated with PyKDL.so. In this way I can use the header file and PyKDL.so when building 
the python module "pycontrollers" (see code above).

MatthijsBurgh, 2020-03-02:
No, the correct answer to my question is: You are creating your own Python Module binded to cpp. Therefore you need PyKDL instead of KDL.

If you are building PyKDL from source, try to add `--sip-h` somewhere in the cmake files. (ref: https://www.riverbankcomputing.com/static/Docs/sip/command_line_tools.html#cmdoption-sip-module-sip-h)
Or take a look at the generate files in the build folder (I think you need to look in build/PyKDL). Maybe there is already a header file there.

But step one, stop including the kdl headers unless you actually use them for something different than the shown code.

I don't have much time to really figure this out.

MatthijsBurgh, 2020-03-03:
The `--sip-h` only works in sip5. PyKDL uses sip4. So that will not work. At the moment it just isn't possible. I am working on a pybind11 version. That probably will have the feature you are looking for. You just need to be patient for that becoming available.

MatthijsBurgh, 2020-05-22:
Also pybind does not generate headers. You could play around with some pybind options for generation of the library etc. I am not sure how this should be done. Good luck

MatthijsBurgh, 2020-06-07:
Closing because of no activity

closed 2020-06-07 by MatthijsBurgh