orocos/orocos_kinematics_dynamics #225: Chocolately package

jacobperron, 2020-03-09:
Greetings, orocos kdl is available upstream in Debian and Ubuntu and being added to rosdistro (https://github.com/ros/rosdistro/pull/23841). I also noticed that it's available via brew (https://formulae.brew.sh/formula/orocos-kdl). For the ROS 2 project, we are currently relying on a [fork](https://github.com/ros2/orocos_kinematics_dynamics) for building and packaging orocos kdl on Windows. Given that orocos kdl is available upstream for our other supported platforms, it would be great to have a version available on Windows. I think this would let us move away from depending on a from-source build of orocos kdl.

This is a call for any interest in publishing / maintaining a [chocolately](https://chocolatey.org/) package for orocos kdl. 

@seanyen, perhaps there is interested from people involved with the ROS on Windows project?

seanyen, 2020-03-12:
@jacobperron Thanks for pinging me. Not sure have you heard of this [project](https://medium.com/@wolfv/ros-on-conda-forge-dca6827ac4b6)? And I have been bootstrapping many system packages on `conda-forge` for Windows build. I also got a sandbox channel https://anaconda.org/ros-playground which has a prototyping ros-eloquent-desktop build, shadows all `xyz_vendor` packages and uses the system packages provided by `conda-forge`. I just want to amplify the potential of `conda-forge` ecosystem. 

And btw, here are two packages of the interested:
https://anaconda.org/conda-forge/python-orocos-kdl
https://anaconda.org/conda-forge/orocos-kdl

Let me know what do you think.
(cc @wolfv)

wolfv, 2020-03-13:
Thanks for pinging me, Sean. 

Yes, indeed, I've been working quite a bit on conda packages for the ROS ecosystem. It would be awesome if it got more attention at OSRF (I also talked quite a bit about our challenges with @tfoote  at ROSCON).
Yesterday we got Gazebo to work on Linux as well, which shows that conda-forge can handle OGRE + QT applications quite well (it was already working since a while on Windows thanks to @seanyen).

Locally I also had builds for RViz and other QT based packages, so we could indeed have a full distribution of ros-*-desktop on conda :)

wolfv, 2021-03-10:
@mganglb yes, that is the correct way.

`-c conda-forge` means you're using the "conda-forge" channel, and is the same as adding it to your configuration (which you were doing with the lines above).

Channels are like PPA's on Ubuntu, or other software package collections.

Glad to hear that it works. We have a lot more packages for conda now as part of the RoboStack initiative: https://medium.com/robostack/cross-platform-conda-packages-for-ros-fa1974fd1de3 (for Windows, OS X and Linux)

MatthijsBurgh, 2025-04-24:
Stale

closed 2025-04-24 by MatthijsBurgh