orocos/orocos_kinematics_dynamics #231: Incompatibility with SIP >= 5, while SIP 4 is deprecated and no longer available in PyPi

pptaszni, 2020-04-12:
I mentioned this problem already in another repo: https://github.com/ros-visualization/rqt/issues/211 but no action taken. On 08.10.2019 SIP 5.0.0 was released and consequently all previous versions were deprecated and are no longer available in PyPi repository for Python3. In `python_orocos_kdl` very old scripts `FindSIP.py` and `SIPMacros.cmake` are used. I am able to modify them a bit by using new `import sipbuild` and modifying the flags, but I still didn't manage to obtain working version of the bindings. Is it possible to address this issue and update the python bindings (doesn't need to be done with SIP, Pybind11 works quite well);

MatthijsBurgh, 2020-04-18:
SIP 5 never supported Python 2. As the master branch of KDL never supported Python 3, this has never been an issue. I am aware that the build process is out of sync, but moving is not a priority right now.
#229 moves to PyBind11 and adds Python 3 support, both PyBind11 and SIP (Still the old build scripts).
Otherwise feel free to start working on a PR to move to SIP 5. If stuff doesn't work, people can contribute, but without a PR, nobody can help.

MatthijsBurgh, 2020-06-15:
Closing due to no activity

closed 2020-06-15 by MatthijsBurgh