orocos/orocos_kinematics_dynamics #232: Build Error during make

marcos452, 2020-04-12:
[image]

hallo,guys .somone know why?

MatthijsBurgh, 2020-04-18:
A little bit more information about your system and used version would be nice. Also a copy of the log instead of an image is much easier.

marcos452, 2020-04-18:
> A little bit more information about your system and used version would be nice. Also a copy of the log instead of an image is much easier.

thanks，i have solved this problem. Have you ever use urdf_praser_py with pykdl to get chain to calculate inverse kinematic?
thanks for your reply

MatthijsBurgh, 2020-04-18:
@marcos452 I have not, I don't have any knowledge of `urdf_parser_py`. To get from urdf to kdl/PyKDL, use [kdl_parser](https://github.com/ros/kdl_parser). As this is unrelated to the original bug. Please close this issue. Any issue regarding getting a tree/chain, should be reported to `kdl_parser`. Any issue regarding solving the tree/chain should be reported here in a new issue.

marcos452, 2020-04-18:
@MatthijsBurgh thanks a lot, i will close the issue. I have some questions about kdl_parser.i will open new issue in kdl_parser. could u help me?

marcos452, 2020-04-18:
@MatthijsBurgh I use the kdl_parser to get chain to calculate inverse kinematic. i input the initial angle and desired frame.i get 0 from q_out. i don't how why?

closed 2020-04-18 by marcos452