orocos/orocos_kinematics_dynamics #233: We have Python type stub file (*.pyi) for PyKDL

mikolajz, 2020-04-15:
Hi,
  To be able to run type checks on our code using PyKDL, we have created a pyi file for PyKDL. This allows mypy to check if we are using it correctly (and makes IDEs like PyCharm autocomplete methods for PyKDL objects). Do you want to include it in your package?

The file is attached, so that you can have a look:
[PyKDL.zip](https://github.com/orocos/orocos_kinematics_dynamics/files/4482388/PyKDL.zip)

MatthijsBurgh, 2020-04-17:
@mikolajz Could you please provide the way you have generated the stub file? I have also implemented this for the SIP based PyKDL, https://github.com/orocos/orocos_kinematics_dynamics/commit/ebb7596a3748a9e0538df9374ba1810af22bbee8. Sadly this only works for SIP>=4.18. (As I am more focused on getting a PyBind11 based PyKDL, I haven't opened a PR for this.)

Did you use the same method to generate the stub file?

P.S. I am planning to add stub generation also to the PyBind11 based PyKDL, but this is also future work.

mikolajz, 2020-04-22:
These stubs were generated by hand (thus, they will be harder to maintain...). Is the PyKDL that one downloads from PIP based on PyBind11 or SIP?

MatthijsBurgh, 2020-04-22:
That one is PyBind11 based and maintained by @zchen24.

jf---, 2020-06-09:
>P.S. I am planning to add stub generation also to the PyBind11 based PyKDL, but this is also future work.

That's a very cool idea...

Might make sense to setup a GH action ( [here's a mypy checker](https://github.com/jpetrucciani/mypy-check) ) such that mypy checks are in place?