orocos/orocos_kinematics_dynamics #242: Build for windows failing due to missing externals

XenophobiusLovegood, 2020-05-14:
When I tried to build the library in Windows 10 using Visual Studio 2019, I get errors like these

`LNK2019	unresolved external symbol "public: __thiscall KDL::JntArray::JntArray(unsigned int)" (??0JntArray@KDL@@QAE@I@Z) referenced in function "public: __thiscall KDL::ChainFdSolver_RNE::ChainFdSolver_RNE(class KDL::Chain const &,class KDL::Vector)" (??0ChainFdSolver_RNE@KDL@@QAE@ABVChain@1@VVector@1@@Z)	A Cpp App	D:\%user%\14thMay\Solution_using_VS\create_dll_repo\dotnet-how-to-create-a-clr-wrapper-of-c-for-using-in-c\A Cpp App\chainfdsolver_recursive_newton_euler.obj	1	
`
`LNK2019	unresolved external symbol "public: __thiscall KDL::JntArray::~JntArray(void)" (??1JntArray@KDL@@QAE@XZ) referenced in function __unwindfunclet$??0ChainFdSolver_RNE@KDL@@QAE@ABVChain@1@VVector@1@@Z$4	A Cpp App	D:\%user%\14thMay\Solution_using_VS\create_dll_repo\dotnet-how-to-create-a-clr-wrapper-of-c-for-using-in-c\A Cpp App\chainfdsolver_recursive_newton_euler.obj	1	
`
This type of error is caused when external libraries are missing.

MatthijsBurgh, 2020-05-14:
I am not sure what you are doing as you are providing very little context to the error.

I assume you are trying to include kdl in the following example, https://bitbucket.org/hintdesk/dotnet-how-to-create-a-clr-wrapper-of-c-for-using-in-c/src/default/.

I assume you are not using the default build procedure. So in that case it is not really our problem. So in case you want any form of support, please provide a good and extended description.

XenophobiusLovegood, 2020-05-14:
My apologies if I did not describe the problem correctly. The directions for building in MS Visual Studio are not indicated, only the Linux make procedure is given. So I am not aware of the default build procedure for Windows 10.

MatthijsBurgh, 2020-05-14:
I think the CMake should work in VS as well.

XenophobiusLovegood, 2020-05-15:
Figured out issue. Missing Eigen3 libraries. Closing issue.

closed 2020-05-15 by XenophobiusLovegood