orocos/orocos_kinematics_dynamics #243: Build for Windows creates only a lib file, no header files

XenophobiusLovegood, 2020-05-15:
Following up to my previous issue #242, I successfully built the KDL library. For the library to be used inside a C# application, a header file needs to be generated, but Visual Studio builds only a lib file. Could anyone assist me a little here regarding the header file generation?

MatthijsBurgh, 2020-05-15:
What do you mean with "no header generation"? Are there any specific header files VS can't find? Because there is a difference between not existing and not knowing where they are.

The header files already exist in the source code,, except for `config.h.in`, which is configured by CMake.

XenophobiusLovegood, 2020-05-15:
I mean this: For a dll to be used at runtime, a header file is needed to be included which exposes all the functions inside. This header file is not being generated by the build system.

MatthijsBurgh, 2020-05-15:
Did you trying running a cpp application first? This again might be related by you trying to include it in c#.

MatthijsBurgh, 2020-05-22:
@XenophobiusLovegood any progress on a cpp app?

XenophobiusLovegood, 2020-05-22:
I was unable to link the library to the application, and I couldn't explore it in the object browser in Visual Studio. Currently changing build configuration parameters to see if I get any results.

MatthijsBurgh, 2020-06-07:
Closing because of no activity

closed 2020-06-07 by MatthijsBurgh