orocos/orocos_kinematics_dynamics #245: How to create a trajectory for PTP motion?

xxxdomi, 2020-05-18:
Hello,

I would like to create a Point-to-Point Movement (Params: PoseStart, PoseEnd, MaxVelocity, MaxAccelleration) for my TCP. 

Do you have an example how to realize that motion (creating a path, defining a velocity profile and creating the trajectory)?

Thanks in advance!

Domi

MatthijsBurgh, 2020-05-18:
So you are looking for an Inverse kinematics/dynamics solver. I don't know if kdl includes one, which takes velocity/acceleration limits into account.. Btw are you talking about limits for velocity/acceleration in joint or Cartesian space?

@smits Does kdl have any solver which takes these constraints into account?

xxxdomi, 2020-05-18:
Hello Matthijs, thanks for your fast response.

I've already implemented an ik-solver (ChainIkSolverPos_NR_JL) and a fw-solver (ChainFkSolverPos_recursive). I was also able to create and execute a LIN-Movement with my robot (Path_Line + VelocityProfile_Trap -> Trajectory_Segment) and that works fine - for the lin-movement, i am talking about the velocity and acceleration in cartesian space, which are used for the velocity profile.

But now, at first, i would like to create a point-to-point movement for my robot - but i don't know how to implement that with the available classes orocos_kdl are providing. unfortunately, i didn't find an example implementation of this issue.

If i am able to create a trajectory for a point-to-point movement, i would like to limit the maximum joint velocities and accelerations. Is that possible with the c++ classes you're providing? 

Best regards,
Dominik

eaertbel, 2020-05-18:
I assume that you want to achieve a joint space point-to-point motion from one joint position to another joint position with joint velocity limits and joint acceleration limits.

This can be realized by using VelocityProfile_Trap for each of the joints, and keeping track of the maximal duration.  In order to ensure that all joints move synchronously (start/stop at the same time), adapt the velocity profiles using VelocityProfile_Trap::setProfileDuration() with the maximal duration.

xxxdomi, 2020-05-18:
Hello Erwin, thank you very much for your response. How would you realize that? If i understand you correctly, you would create a path and a velocity-profile for each axis, wouldn't you?

As i've already mentioned, i would like to define a start-frame and an end-frame - additional to that i would like to define the max (angular) velocities and accelerations of each joint - and move the robot (tcp) as fast as possible from start frame to the end frame.

May you provide a code snippet how you would create the path, velocity-profile and the trajectory or describe in more detail, how you would realize that? 

Thanks in advance, 
Dominik

eaertbel, 2020-05-18:
No, for a PTP motion you do not create a path.

The PTP motion of a typical industrial robot controller uses something similar as:
 - inverse position kinematics of end-frame to determine the end pose in joint space coordinates.
    (preferably the joint space solution that is closest to the starting pose, or the solution with the same configuration as the starting pose, if the robot has joints with a range > 360, taking also into account the "windings') 
 - directly interpolating the joints using VelicityProfile_Trap routines between start pose in joint space coordinates and end pose in joint space coordinates.  (scaled such that they all have the same time-duration as explained above).

xxxdomi, 2020-05-20:
Thanks for your helpful answer - i could test it and it worked partially quite good. But now I'm facing a problem which I think you tried to describe. If I move my robot a few times from poseA to poseB and back from B to A (by a point-to-point motion), the inverse kinematics solver determines another solution in joint space for the end pose each time. 

Is there a way that the inverse solver (in my case the "ChainIkSolverPos_NR_JL") gives me a solution in a preferred joint space? Or how would you realize that?

Thanks in advance. Best regards from Austria!

MatthijsBurgh, 2020-06-27:
@eaertbel thansk for your help. @xxxdomi I hope you find a solution. Closing due to inactivity.

closed 2020-06-27 by MatthijsBurgh