orocos/orocos_kinematics_dynamics #249: PyKDL throwing segmentation fault after passing a str to Tree() method in ROS - Noetic

Murtaza786-oss, 2020-05-26:
Just installed ROS-Noetic and ran into this bug I suspect
[image]

It seems that PyKDL.Tree() method is not working because of a segmentation fault.
This error never used to occur on ROS/Melodic with Python 2.7. Can someone please provide a solution?

This is the package.xml from the python_orocos_kdl currently installed in ros-noetic,

[image]

Also, tried to build this repo  from source. But  landed up with these errors can you please help me?

[image]

MatthijsBurgh, 2020-05-26:
Did you initialize the submodule?

Murtaza786-oss, 2020-05-26:
I am not sure how should I initialize it can you please explain it to me briefly?

MatthijsBurgh, 2020-05-26:
`git submodule update --init`

Murtaza786-oss, 2020-05-26:
It gives me this error 

`fatal: not a git repository (or any of the parent directories): .git`

MatthijsBurgh, 2020-05-26:
Did you do it in the git folder?

Murtaza786-oss, 2020-05-26:
ok thank you, I have successfully generated the PyKDL.so file which is in the path `/home/murtaza/catkin_ws3/devel_isolated/python_orocos_kdl/lib/python3.8/dist-packages/PyKDL.so`

but when I try to import PyKDL after compiling python still refers to the old path that is `/usr/lib/python3/dist-packages/PyKDL.cpython-38-x86_64-linux-gnu.so`. How can I alter this?

MatthijsBurgh, 2020-05-26:
Add `/home/murtaza/catkin_ws3/devel_isolated/python_orocos_kdl/lib/python3.8/dist-packages` to your PYTHONPATH environment variable

Murtaza786-oss, 2020-05-26:
I used catkin_make_isolated because I was getting the following error with catkin_make,

```
-- ~~  traversing 2 packages in topological order:
-- ~~  - orocos_kdl (plain cmake)
-- ~~  - python_orocos_kdl (plain cmake)
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
CMake Error at /opt/ros/noetic/share/catkin/cmake/catkin_workspace.cmake:100 (message):
  This workspace contains non-catkin packages in it, and catkin cannot build
  a non-homogeneous workspace without isolation.  Try the
  'catkin_make_isolated' command instead.
Call Stack (most recent call first):
  CMakeLists.txt:67 (catkin_workspace)

-- Configuring incomplete, errors occurred!
See also "/home/murtaza/catkin_ws3/build/CMakeFiles/CMakeOutput.log".
See also "/home/murtaza/catkin_ws3/build/CMakeFiles/CMakeError.log".
make: *** [Makefile:320: cmake_check_build_system] Error 1
```

After adding to PYTHONPATH I get the following error,
`ImportError: /home/murtaza/catkin_ws3/devel_isolated/python_orocos_kdl/lib/python3.8/dist-packages/PyKDL.so: undefined symbol: _ZNK3KDL6Vector4NormEd`

MatthijsBurgh, 2020-05-26:
Try using python catkin tools

Murtaza786-oss, 2020-05-26:
So I installed the catkin tools package but it seems that it is supporting only py2 because I get syntax error when I try running catkin build like,

```
Traceback (most recent call last):
  File "/usr/local/bin/catkin", line 5, in <module>
    from catkin_tools.commands.catkin import main
  File "/usr/local/lib/python3.8/dist-packages/catkin_tools/commands/catkin.py", line 28, in <module>
    from catkin_tools.common import is_tty
  File "/usr/local/lib/python3.8/dist-packages/catkin_tools/common.py", line 23, in <module>
    import trollius as asyncio
  File "/usr/local/lib/python3.8/dist-packages/trollius/__init__.py", line 21, in <module>
    from .base_events import *
  File "/usr/local/lib/python3.8/dist-packages/trollius/base_events.py", line 42, in <module>
    from . import tasks
  File "/usr/local/lib/python3.8/dist-packages/trollius/tasks.py", line 565
    def async(coro_or_future, loop=None):
        ^
SyntaxError: invalid syntax
```

MatthijsBurgh, 2020-05-26:
It should work, but try indeed the python2 version of python catkin tools. You can still build PyKDL on python3.

Murtaza786-oss, 2020-05-27:
Hello so I compiled the workspace using catkin build by using the building the catkin tools from src from this repo [(https://github.com/catkin/catkin_tools.git)]. I get the following error when I compile.

[image]

MatthijsBurgh, 2020-05-27:
Yes it is a warning, not an error. So no problems.

I see that the build_depend doesn't contain a condition. This should be fixed.

Also rosdistro has not specified any option for focal. This should be fixed.

MatthijsBurgh, 2020-05-27:
@Murtaza786-oss The missing SIP shouldn't be a problem, because if you don't set `BUILD_PYKDL_PYBIND11`, it defaults to ON, so PyBind11 is used.

Could you confirm that you don't have any segfaults anymore?

Murtaza786-oss, 2020-05-27:
yes, no segmentation faults anymore thank you for the support. It works fine now!

MatthijsBurgh, 2020-05-27:
Problem solved, closing

closed 2020-05-27 by MatthijsBurgh

YeeCY, 2021-01-14:
> So I installed the catkin tools package but it seems that it is supporting only py2 because I get syntax error when I try running`catkin build`like,
>
> ```
> Traceback (most recent call last):
>   File [...]

I got the same problem, how to fix that?

MatthijsBurgh, 2021-01-14:
> It should work, but try indeed the python2 version of python catkin tools. You can still build PyKDL on python3.

@YeeCY Same problem, same solution ;)

`catkin_tools` is still broken on python3, (https://github.com/catkin/catkin_tools/issues/594)

YeeCY, 2021-01-14:
> > It should work, but try indeed the python2 version of python catkin tools. You can still build PyKDL on python3.
> 
> @YeeCY Same problem, same solution ;)
> 
> `catkin_tools` is still broken on python3, ([catkin/catkin_tools#594](https://github.com/catkin/catkin_tools/issues/594))

Okay, so currently we can only `catkin_make` this repo with other packages in a workspace with python2, right?

And I choose to build the repo and install the headers into the system path without catkin, it works as well.

MatthijsBurgh, 2021-01-14:
You can use `catkin_tools`. Out of the box only the python2 version of `catkin_tools` does work. With extra steps you can get the python3 version to work as well.

Both the python2 and python3 version of `catkin_tools` can build PyKDL for python2 and python3.

And indeed you use the plain `cmake/make` install procedure. Though this requires a manual update, while catkin will rebuild it automatically.

YeeCY, 2021-01-14:
> You can use `catkin_tools`. Out of the box only the python2 version of `catkin_tools` does work. With extra steps you can get the python3 version to work as well.
>
> Both the python2 and python3 [...]

Got it