orocos/orocos_kinematics_dynamics #251: make artifacts relocatable

dirk-thomas, 2020-06-02:
By avoiding to embed the absolute path of the install prefix in e.g. the .pc file. Similar to ros2/urdfdom#20.

Related to ros2/ros2#606.

MatthijsBurgh, 2020-06-05:
@dirk-thomas would this be solved by copying the changes from https://github.com/ros2/urdfdom/pull/20? Or does it require more changes? I.E.  
copying the following line as well?https://github.com/ros2/urdfdom/blob/01185a3a5ba5e41001bedc867f2baf3ce3242dec/cmake/pkgconfig/urdfdom.pc.in#L4

I am not that experienced with CMake/PkgConfig.

meyerj, 2020-06-05:
I already have a related patch in the backlog to cleanup the relative paths in [KDLConfig.cmake.in](https://github.com/orocos/orocos_kinematics_dynamics/blob/0bc547a7922ba842d0bd2cba60c4878b19d15b8e/orocos_kdl/KDLConfig.cmake.in#L8):

https://github.com/orocos/orocos_kinematics_dynamics/blob/0bc547a7922ba842d0bd2cba60c4878b19d15b8e/orocos_kdl/KDLConfig.cmake.in#L8

Okay, only loosely related: Using `../../..` in the exposed `INCLUDE_DIRS` adds the standard `include` path not specific to KDL at least twice to the compiler command-line, because it is not recognized as a duplicate, and causes problems with include directory ordering or with treating include directories of upstream packages as system include dirs.

Furthermore forwarding include directories of dependencies like this is not nice and can also break relocation. Instead, `KDLConfig.cmake` should find `Boost` and `Eigen` again with `find_package()`, and not hard-code the include directories found at compile-time.

dirk-thomas, 2020-06-05:
> would this be solved by copying the changes from ros2/urdfdom#20? Or does it require more changes?Your `.pc` file: https://github.com/orocos/orocos_kinematics_dynamics/blob/0bc547a7922ba842d0bd2cba60c4878b19d15b8e/orocos_kdl/src/kdl.pc.in#L1-L10

only uses `prefix` to determine the variables `libdir` / `includedir` which are then used to define the properties `Libs` and `Cflags`. So only the first line defining `prefix=@CMAKE_INSTALL_PREFIX@` needs to be changed to not contain an absolute path.

dirk-thomas, 2020-06-05:
> would this be solved by copying the changes from ros2/urdfdom#20? Or does it require more changes?

Your `.pc` file: https://github.com/orocos/orocos_kinematics_dynamics/blob/0bc547a7922ba842d0bd2cba60c4878b19d15b8e/orocos_kdl/src/kdl.pc.in#L1-L10

only uses `prefix` to determine the variables `libdir` / `includedir` which are then used to define the properties `Libs` and `Cflags`. So only the first line defining `prefix=@CMAKE_INSTALL_PREFIX@` needs to be changed to not contain an absolute path.

dirk-thomas, 2020-06-05:
> would this be solved by copying the changes from ros2/urdfdom#20? Or does it require more changes?

Your `.pc` file: https://github.com/orocos/orocos_kinematics_dynamics/blob/0bc547a7922ba842d0bd2cba60c4878b19d15b8e/orocos_kdl/src/kdl.pc.in#L1-L10

only uses `prefix` to determine the variables `libdir` / `includedir` which are then used to define the properties `Libs` and `Cflags`. So only the first line defining `prefix=@CMAKE_INSTALL_PREFIX@` needs to be changed to not contain an absolute path.

> I.E.
copying the following line as well?

That depends if you know the relative path the `.pc` is being installed to. In `urdfdom` that path is "dynamic" so we need to pass it in. If your CMake logic always installs the `.pc` file into `lib` then there is no need to change the current line: `libdir=${prefix}/lib`

PR #255 (2020-06-07): Make .pc file relocatable and refactor cmake config
Fixes #251:
> By avoiding to embed the absolute path of the install prefix in e.g. the .pc file. Similar to ros2/urdfdom#20.

and also addresses https://github.com/orocos/orocos_kinematics_dynamics/issues/251#issuecomment-639401124:
> Using `../../..` in the exposed `INCLUDE_DIRS` adds the standard `include` path not specific to KDL at least twice to the compiler command-line, because it is not recognized as a duplicate, and causes problems with include directory ordering or with treating include directories of upstream packages as system inc [... cut]
commit a540506e (2020-06-07): Make pkg-config file relocatable and move template to the package root (fix #251)

${pcfiledir} refers to the installed location of the .pc file.
Added @LIB_SUFFIX@ to libdir to match install destination in src/CMakeLists.txt.
CMake variable KDL_INCLUDE_DIRS was unused.
closed 2020-06-07 by meyerj