orocos/orocos_kinematics_dynamics #260: Build error on Arch Linux

b-reich, 2020-06-13:
I am trying to build python_orocos_kdl for ROS Noetic. It wont build on Arch with SIP Version 4.19.23.

Its also reported [here](https://github.com/ros-noetic-arch/ros-noetic-python-orocos-kdl/issues/1)

MatthijsBurgh, 2020-06-13:
I have tested SIP 4.19.23 on Ubuntu 18.04 Bionic. That gives me the same errors. SIP 4.19.22 does work for me. The problem is therefore the SIP release. Please debug yourself which change between 4.19.22 and 4.19.23 is causing the error. You can contact the developer of SIP via the PyQt mailing list, https://www.riverbankcomputing.com/mailman/listinfo/pyqt.

jspricke, 2020-06-26:
I looked into this as I have the same problem on Debian: https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=963296
sip 4.9.23 started to generate a copy helper more often: https://www.riverbankcomputing.com/hg/sip/rev/22ababb6d14a#l3.63 that seems to need a copy assignment operator.
The problem is that the ChainDynParam class (and all the others) has a reference member and the compiler deletes the default assignment operator because it wouldn't work. According to https://www.geeksforgeeks.org/default-assignment-operator-and-references/ you have to define a assignment operator.

MatthijsBurgh, 2020-06-26:
So cpp does generate an copy constructor, but not an assignment operator in case a class has a reference member.

I think it is incorrect that SIP assumes that a correct assignment operator is available if a copy constructor is available. As cpp does generate a default copy constructor in case the class has a reference member. But no default assignment operator is created. So SIP is now forcing everybody to implement this.

I would prefer SIP removing these constraints.

jspricke, 2020-06-26:
> I would prefer SIP removing these constraints.

I guess that's a little outside of what the parser is capable of (as far as I understood it). 
The comment in the code says they assume that the assingment operator is there: https://www.riverbankcomputing.com/hg/sip/rev/22ababb6d14a#l1.48

Anyway, I've send a mail to the pyqt list (will add a link once it's in the archive).

MatthijsBurgh, 2020-06-26:
I noticed the comment. I think that assumption applies too much restrictions for the user. Also it is a significant change compared to the old situation.

People experience the similar problem with PyQt, http://python.6.x6.nabble.com/PyQt-5-13-2-issues-with-sip-4-19-23-td5276550.html

(@jspricke did you subscribe to the pyqt mailing list? It is needed to be able to send messages. Normally it is added in less than 5 minutes.)

jspricke, 2020-06-26:
Yeah, seems like one need to be subscribed to post it's there now: https://www.riverbankcomputing.com/pipermail/pyqt/2020-June/043025.html

MatthijsBurgh, 2020-06-26:
@jspricke please check if the errors for `ChainDynParam` disappear with my fix from #269

jspricke, 2020-06-26:
Nop, similar to what I posted here: https://www.riverbankcomputing.com/pipermail/pyqt/2020-June/043027.html

closed 2020-06-26 by MatthijsBurgh

ninafiona, 2021-02-23:
I am getting the same build errors, is this fixed now? If not, where can I get download the older version of sip?

PR #270 (2020-06-26): Declare assignment operator private for SIP
Starting with v4.19.23 SIP expects a working operator= or one marked
private explicitly. All classes in this PR have a reference member
(&chain) resulting in the compiler deleting the default assignment
operator. This PR makes this known to SIP as well.

Fixes #260
commit f8d488ca (2020-06-26): Declare assignment operator private for SIP (Closes: #260)

Starting with v4.19.23 SIP expects a working operator= or one marked
private explicitly. All classes in this PR have a reference member
(&chain) resulting in the compiler deleting the default assignment
operator. This PR makes this known to SIP as well.