orocos/orocos_kinematics_dynamics #263: Newest version of orocos kdl

DimitriEckert137, 2020-06-17:
Hi everyone

I tried to use the inverse dynamics solver for trees (documentation) from the kinematis and dynamics library (kdl). I have installed the newest version of kdl with :

sudo apt install ros-melodic-orocos-kdl

So it should be the newest version (1.4.0-0bionic.20200303.062327).

But when i looked at the folder in /opt/ros/melodic/include/kdl there was no treeidsolver_recursive_newton_euler.hpp file, although according to the above mentioned documentation and this github repository it should be there. I managed to do a workaround by simply copying the necessary files into the folder manually and adapting some others so they are compatible and it somehow works. But i guess this is a horrible way to do it and i am not sure if it is not going to break down when i try different stuff. I get some seg fault issues which i am not sure if they are linked to this patchy approach. 

Can anyone tell me how i would do this the "right" way?

I also tried to create an account on the orocos webpage but when i responded to the invitation email like required i got an "Undelivered Mail Returned to Sender". Is this website and the orocos project even running anymore?

(i am using ubuntu 18.04 and ros melodic)

MatthijsBurgh, 2020-06-18:
The released version in Melodic is `1.4.0`, see the `1.4.0` tag, https://github.com/orocos/orocos_kinematics_dynamics/releases/tag/v1.4.0.

The current master still show 1.4.0, as it has not been released, the version is not bumped yet.

You can compile it from source. Just clone the repo in your workspace, make sure all dependencies are installed and use catkin.

DimitriEckert137, 2020-06-19:
Ok. So I cloned the repo into my catkin_ws/src folder. Then I ran 
`catkin build orocos_kdl `
Which worked fine.
Then I ran
`rosdep install orocos_kdl`
Which also worked fine (I can't find the name of the additional stuff that was installed though since I closed the terminal in the meantime. Is there a way to see what rosdep has installed?)
But now when I want to run my own package I get the following error:
```
QObject::connect: Cannot queue arguments of type 'QVector<int>'
(Make sure 'QVector<int>' is registered using qRegisterMetaType().)
QObject::connect: Cannot queue arguments of type 'QVector<int>'
(Make sure 'QVector<int>' is registered using qRegisterMetaType().)
/opt/ros/melodic/lib/rviz/rviz: symbol lookup error: /opt/ros/melodic/lib/libmoveit_dynamics_solver.so.1.0.4: undefined symbol: _ZNK3KDL6Vector4NormEv

```
After which rviz dies. I tried it without including any of the kdl stuff but the error still appeared. When I run rviz on its own with default configuration It works fine.
I looked into the launch file of my node I found this part:
```
 <!-- launch Rviz with MoveIt -->
    <include file="$(find ur3_e_moveit_config)/launch/moveit_rviz.launch">
      <arg name="config" value="true"/>
    </include>
```
The moveit_rviz.launch file looks like this:

```
<launch>

  <arg name="debug" default="false" />
  <arg unless="$(arg debug)" name="launch_prefix" value="" />
  <arg     if="$(arg debug)" name="launch_prefix" value="gdb --ex run --args" />

  <arg name="config" default="false" />
  <arg unless="$(arg config)" name="command_args" value="" />
  <arg     if="$(arg config)" name="command_args" value="-d $(find ur3_e_moveit_config)/launch/moveit.rviz" />
  
  <node name="$(anon rviz)" launch-prefix="$(arg launch_prefix)" pkg="rviz" type="rviz" respawn="false"
	args="$(arg command_args)" output="screen">
    <rosparam command="load" file="$(find ur3_e_moveit_config)/config/kinematics.yaml"/>
  </node>

</launch>
```

The loaded kinematics.yaml file:
```
manipulator:
  kinematics_solver: kdl_kinematics_plugin/KDLKinematicsPlugin
  kinematics_solver_search_resolution: 0.005
  kinematics_solver_timeout: 0.005
  kinematics_solver_attempts: 3
``` 

So maybe there is a conflict since rviz uses this kdl_kinematics_plugin and now with the orocos_kdl package installed it doesn't work anymore?
I honestly wouldn't know where to start looking for this 'QVector<int>' since the rviz part was not written by me.

MatthijsBurgh, 2020-06-19:
I assume you have loaded your workspaces correct. In that case, this does show that other software isn't compatible with the current master. So you can't use that. You will need to remove all related files, source and compiled files.

DimitriEckert137, 2020-06-19:
What exactly do you mean by "loading the workspace"? 
So you think I should try to completely get rid of the kdl_kinematic_plugin ?

MatthijsBurgh, 2020-06-19:
No, get rid of `orocos_kdl` in your workspace. Use the system version

You need to source the `setup.bash` of your workspace to be available in your terminal. That is what I meant with loading your workspace.

DimitriEckert137, 2020-06-19:
System version being the one I get by  `sudo apt install ros-melodic-orocos-kdl` ? 
In this version there is no tree inverse dynamics solver right?

Ok yes, i sourced it.

MatthijsBurgh, 2020-06-19:
Yes that is the system version.
Yes, there is no `Tree inverse dynamics solver` in that version.

If want to use both the `Tree inverse dynamics solver` and other libraries, which the system versions are not compatible with the newest kdl. You will need to compile those libraries from source.

DimitriEckert137, 2020-06-19:
Ok thank you very much. So I will switch t the system version and will try to convert my tree to a chain and use the chain id solver.

closed 2020-06-19 by MatthijsBurgh