orocos/orocos_kinematics_dynamics #264: Segfault in treeidsolver_recursive_newton_euler.cpp

DimitriEckert137, 2020-06-18:
@smits 

Hi everyone 

This error might be due to a bad installation (which i mentioned in [this issue](https://github.com/orocos/orocos_kinematics_dynamics/issues/263#issue-640485736) which i am waiting for an answer to) bu I think its a different problem. I am working with a simulated ur3e robot. I use the kdl_parser to to get a kdl tree from the urdf file. 

Its a segfault and by adding printf statement to treeidsolver_recursive_newton_euler.cpp I could narrow it down, so I am pretty sure it happens on the following line:
` const std::string& parname = GetTreeElementParent(segment->second)->first;`
when it passes it the SECOND time:
```
Thread 8 "continuous_offs" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fffdeffd700 (LWP 12010)]
0x00007ffff7bb830c in std::_Rb_tree_const_iterator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, KDL::TreeElement> >::operator->() const () from /home/dimitri/catkin_ws/devel/lib/libcontinuous_offset_calib.so
```
I tried to printf the name of the current segment to see at which segment the algorithm is at that moment but this also leads to a segfault (already at the first time it passes it). 
```
Thread 8 "continuous_offs" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fffe6ffd700 (LWP 13207)]
0x00007ffff683f470 in std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::c_str() const ()
   from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
```
This is my code:

ContinuousOffsetCalib.hpp:

```

#pragma once

#include <any_node/any_node.hpp>
#include <UrKdl.hpp>
#include <sensor_msgs/JointState.h>
#include <geometry_msgs/WrenchStamped.h>
#include <std_msgs/String.h>
#include <std_msgs/Float64.h>
#include <sstream>
#include <memory>    // for std::shared_ptr

namespace continuous_offset_calib {

class ContinuousOffsetCalib : public any_node::Node {

public:

  ContinuousOffsetCalib() = delete;
  explicit ContinuousOffsetCalib(any_node::Node::NodeHandlePtr nh) : any_node::Node(nh) {}
  ~ContinuousOffsetCalib() override = default;

  bool init() override;
  void cleanup() override;
  bool update(const any_worker::WorkerEvent& event);

  void jointStatesSubscriberCallback(const boost::shared_ptr<const sensor_msgs::JointState>& msg);
  void rokuSubscriberCallback(const boost::shared_ptr<const geometry_msgs::WrenchStamped>& msg);

  void preCleanup() override;

  ros::Subscriber joint_states_subscriber;
  ros::Subscriber roku_subscriber;
  ros::Publisher filtered_FT_publisher;

private:

  ur_kdl::UrKdl complete_robot_tree;

  KDL::JntArray current_joint_positions;            // current joint positions              

  KDL::JntArray current_joint_velocities;            // current joint positions              

  struct joint_torques {
[... 156 lines cut ...]
  sensor_msgs::JointState calculateTorques(const KDL::JntArray &q, const KDL::JntArray &q_dot, const KDL::JntArray &q_dotdot, const KDL::WrenchMap& f_ext, KDL::JntArray &tau){
    int error = 0;
    MELO_INFO("before carttojnt tau: %i", tau.rows());
    error = invDynSolver->CartToJnt(q, q_dot, q_dotdot, f_ext, tau);
    MELO_INFO("after carttojnt tau: %i", tau.rows());
    sensor_msgs::JointState torques;
    std::vector<double> effort;
    MELO_INFO("calculateTorques, %i", error);
    for(int i = 0; i < tau.data.rows(); i++){
      effort.push_back(tau.data[i]);
      MELO_INFO("%i   %f", i, tau.data[i]);
    }
    torques.effort = effort;
    return torques;
  }

private:

  KDL::WrenchMap current_wrench_FTsensor;
  KDL::TreeIdSolver_RNE* invDynSolver;

};

} // namespace ur_kdl
```

MatthijsBurgh, 2020-06-18:
Can you please split the line:
```cpp
const std::string& parname = GetTreeElementParent(segment->second)->first;
```
into seperate lines. So we know which part of this line is the problem.
(Maybe add a print inbetween each line)

Can you tell me if the library is compiled with `KDL_USE_NEW_TREE_INTERFACE ` on or off?

DimitriEckert137, 2020-06-18:
Ok i split it up like this:
```
      const SegmentMap::const_iterator& parentseg =  GetTreeElementParent(segment->second);
      const std::string& parname = parentseg->first;
```
and it seems like the error occurs in the second line.

For the CMakeCache: I dont find this file. Which is probably due to the installation problems I mentioned above ([here](https://github.com/orocos/orocos_kinematics_dynamics/issues/263) again the link to the github issue). So this might be the problem after all.

MatthijsBurgh, 2020-06-18:
CMakeCache is in the build folder (`catkin_ws/build/orocos_kdl`).

So have you compiled kdl from source or are you using `ros-$ROS_DISTRO-orocos-kdl`??

It is very logical that the second line can fail, as it is an iterator, which can point to the end of an iterable. In that case it is an nullptr. So could you add a check?
```
if (parentseg)
    std::cout << "Iterator is valid" << std::endl;
else
  std::cout << "iterator is invalid" << std::endl;
```

MatthijsBurgh, 2020-06-24:
It has not been concluded yet if this bug is related to your implementation or should be fixed in the library.

DimitriEckert137, 2020-06-24:
> It has not been concluded yet if this bug is related to your implementation or should be fixed in the library.

ok. I just closed it because i dont follow this approach anymore and use the chain inverse dynamics solver instead.

francofusco, 2020-11-12:
Hey! I'm the author of the solver, sorry for not having seen this issue before... 😞 

My best guess is that the problem is you build the `KDL::Tree` instance as a temporary in the constructor of `UrKdl`. The ID solver stores a reference to a tree object, [see here](https://github.com/orocos/orocos_kinematics_dynamics/blob/f138e21db410d5eaef448a8ab270df92d76bed5c/orocos_kdl/src/treeidsolver_recursive_newton_euler.hpp#L72). I imagine that after construction of the `UrKdl` object, the tree instance is destroyed and you get undefined behavior. Could you try to add the tree as a member variable of `UrKdl` and see if this solves the issue? If not, I'll dig more into the issue and try to find the bug :+1: