orocos/orocos_kinematics_dynamics #268: Problems with FKSolver if kinematic structure if transformation matrix is constructed by DH_CRAIG1989

easUser, 2020-06-26:
Hello community,

a few days ago i found your repo and tested the FW and IK solver with the iiwa model you've provided with the c++ code - everything seems to work.

After i tried to test the solvers with some other roboters (f.e. kuka kr15-2, ...) i recognized that the initial positions of the tcp weren't correct. So, in short, if i describe the kinematics with dh parameter by craig the fw kinematics solver doesn't work as expected.

I provided you an example below - i'am not sure if i am missunderstanding something ...

I took an example robot from craig's "introduction to robotics" book.

[image]

The dh parameters are shown in the picture below (Notice: The final offset for the end effector isn't considered; All joints are rotational; the z-axis are pointing out of the paper).

[image]

In the following code you will see how the robot kinematics are described with dh parameters. After i initialized the FW-KIN Solver i set initial joint values and try to solve the forward kinematics.

```
	using namespace KDL;
	using namespace std;

	double PI_Const = 3.14159265359;

	double l1 = 1;
	double l2 = 2;

	Chain chain1;

	//joint 1
	chain1.addSegment(KDL::Segment(Joint(Joint::RotZ),
		Frame::DH_Craig1989(0.0, 0.0, 0.0, 0.0)));

	//joint 2
	chain1.addSegment(KDL::Segment(Joint(Joint::RotZ),
		Frame::DH_Craig1989(l1, 0, 0.0, 0.0)));

	//joint 3
	chain1.addSegment(KDL::Segment(Joint(Joint::RotZ),
		Frame::DH_Craig1989(l2, 0.0, 0.0, 0.0)));

	int n = chain1.getNrOfJoints();

	ChainFkSolverPos_recursive fwdkin(chain1);

	JntArray q_init(n);
	JntArray q_sol(n);

// 1. attempt: Sets a position for the three joints in rad
	q_init(0) = 0;
	q_init(1) = 0;
	q_init(2) = 0;

	Frame pos_goal, pos_reached;
	fwdkin.JntToCart(q_init, pos_goal);
	**// TCP - Analytic calculated (target) pos: TCP_X = 2 + 1 = 3
	// TCP - CurrentPos: TCP_X = 2 + 1 = 3 -> CORRECT**

// 2. attempt: Sets a position for the three joints in rad
	q_init(0) = PI_Const * 0.5;
	q_init(1) = 0;
	q_init(2) = 0;

	fwdkin.JntToCart(q_init, pos_goal);
	**// TCP - Analytic calculated (target) pos: TCP_X = 0; TCP_Y = 2 + 1 = 3
	// TCP - CurrentPos: TCP_X = 0; TCP_Y = 3 -> CORRECT**

// 3. attepmt: Sets a position for the three joints in rad
	q_init(0) = PI_Const * 0.5;
	q_init(1) = -PI_Const * 0.5;
	q_init(2) = 0;

	fwdkin.JntToCart(q_init, pos_goal);
	**// TCP - Analytic calculated (target) pos: TCP_X = 1; TCP_Y = 2
	// TCP - CurrentPos: TCP_X = 3; TCP_Y = 0 -> INCORRECT**
```

Why is the last tcp position not correct?

Thanks in advance,
Jimmy

MatthijsBurgh, 2020-06-27:
@smits @meyerj could you handle this one?

smits, 2020-12-21:
I'll take a look when I find some time.

superseppl, 2023-02-06:
I have the same issue on the UR5 robot. When i parse the URDF xacro file the calculation of the end effector postion with the fk_solver seems to be incorrect. 

Current:
[image]

Desired (from https://bjpcjp.github.io/pdfs/robotics/MR_ch04_fwd_kinematics.pdf):
[image]

superseppl, 2023-02-06:
With the library I read my URDF xacro file of the UR5 (https://github.com/ros-industrial/universal_robot/tree/melodic-devel/ur_description/urdf), build the kinematic chain and then calculate the forward kinematics of my end-effector wrt. the world frame.

MatthijsBurgh, 2023-02-06:
Just start very simple.

Make sure you publish the joint states. Run a `robot_state_publisher` and check the TF frames and robot model in rviz. Please share the result. Is that what you expect?

superseppl, 2023-02-07:
@MatthijsBurgh Alright! Here is the setup of the UR5 in zero configuration using the URDF according to REP-199. The slightly thicker frame at the flange is the forward kinematic solution (flange-base_link_inertia) calculated with KDL. 

[image]

MatthijsBurgh, 2023-02-07:
So the tf frames do match the REP-199? (Remember REP-199 is not an official REP)

But the KDL FK solution does have the correct position, but incorrect orientation?

Can you check the frames in between as well?

superseppl, 2023-02-07:
Yes, it has the correct position but incorrect orientation. In the zero configuration, where all joints are 0, the orientation should be [ 0.7071068, 0, 0, 0.7071068 ] in Quaternion [x, y, z, w] according to the [Tool](https://www.universal-robots.com/articles/ur/application-installation/dh-parameters-for-calculations-of-kinematics-and-dynamics/) but it is [0, 0, 0.707, 0.707]. Which looks like:
for the flange:
[image]

for the wrist_3_link:
[image]

MatthijsBurgh, 2023-02-07:
When you flip `wrist_3_link` to the correct orientation, will the `flange` also be correct?

superseppl, 2023-02-08:
Found the issue, we further used Eigen and I was not aware that KDL::Rotation is row-major, while default Eigen::Matrix is column-major.

MatthijsBurgh, 2023-02-08:
@superseppl so no bugs in KDL? So we can close this one?

superseppl, 2023-02-10:
Yes, no bugs, thanks for the support :+1:

closed 2023-02-10 by MatthijsBurgh