orocos/orocos_kinematics_dynamics #271: Syntax Improvement ideas

ahoarau, 2020-06-29:
Hello,
I'm glad to see this great library being updated, so I'm giving my two cents on some improvements that could be made on the syntax, based on my experience:

# KDL::Vector
* 1a. Make the data an Eigen::Vector3d so we can do math with it
* 1b. Make x,y,z accessors public so we can do `v.x = 1.0`, instead of `v.x(1.0)` and `double x = v.x()`

# KDL::Rotation
* Make the data an Eigen::Matrix3d so we can do eigen math with it

# KDL::JntArray
* Make a zero-alloc constructor from an `Eigen::VectorXd` and `std::vector<double>`
* Have an accessor `operator []`
* make it compatible with c++11 range-based iterators
* Make it inherit from a `KDL::Array` class that could be used to create some gains for example. Right now people are either mixing eigen/kdl types or using JntArrays for gains.

# KDL::JntArrayVel
* Make a zero-alloc constructor from eigen, std and JntArrays

# All dynamic size arrays
* Have the ability to "lock" the size of the array and throw exception if some aliasing occurs

What do you guys think ?

meyerj, 2020-11-11:
All sounds good and reasonable. But likely this requires a bigger refactoring and also a bump to version 2, because most of the suggested changes break the API and require changes in user code, too.

But there is definitely a need for such a breaking refactoring, also to iron out many inconsistencies in the current API and to make use of some C++11 or newer features. Maybe we can use this issue to collect more ideas and suggestions, or just to reference it in other issues where API-breaking suggestions have been or will be made.

ahoarau, 2020-11-11:
Is there any roadmap for this ?