orocos/orocos_kinematics_dynamics #272: PyKDL install via pip fails

kunaltyagi, 2020-07-02:
```bash
$ pip install PyKDL --no-cache-dir
Collecting PyKDL
  Downloading PyKDL-1.4.0.tar.gz (1.5 kB)
    ERROR: Command errored out with exit status 1:
     command: /home/kt/virtualenv/noetic/bin/python3 -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-i4evfy2b/PyKDL/setup.py'"'"'; __file__='"'"'/tmp/pip-install-i4evfy2b/PyKDL/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base /tmp/pip-pip-egg-info-dm5k06yb
         cwd: /tmp/pip-install-i4evfy2b/PyKDL/
    Complete output (7 lines):
    running egg_info
    creating /tmp/pip-pip-egg-info-dm5k06yb/PyKDL.egg-info
    writing /tmp/pip-pip-egg-info-dm5k06yb/PyKDL.egg-info/PKG-INFO
    writing dependency_links to /tmp/pip-pip-egg-info-dm5k06yb/PyKDL.egg-info/dependency_links.txt
    writing top-level names to /tmp/pip-pip-egg-info-dm5k06yb/PyKDL.egg-info/top_level.txt
    writing manifest file '/tmp/pip-pip-egg-info-dm5k06yb/PyKDL.egg-info/SOURCES.txt'
    error: package directory 'PyKDL' does not exist
    ----------------------------------------
ERROR: Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.
```

Not using virtualenv doesn't make any difference. It seems to be be an issue in the `setup.py` instructions

MatthijsBurgh, 2020-07-02:
PyKDL on pip is from this repo. It is maintained by @zchen24. It looks it has not been updated in a while. So I wouldn't recommend it.

closed 2020-07-02 by MatthijsBurgh

lucasw, 2021-12-01:
I was also wanting to run ros inside a virtualenv and encountering this error, I ended up building this repo inside my catkin_ws because the virtualenv will use that (but not anything in `/usr/lib/python3/dist-packages`- if PyKDL were instead installed into `/opt/ros/noetic/lib/python3/dist-packages` it would have worked)

   git clone git@github.com:orocos/orocos_kinematics_dynamics
   cd orocos_kinematics_dynamics/python_orocos_kdl
   git submodule update --init

    catkin build

The venv can then make use of every python package inside the `catkin_ws/devel/lib/python3/dist-packages` (and apparently anything else on PYTHONPATH), but rospkg and defusedxml are similar in that they are installed to `/usr/lib`, but pip install inside the venv does work for them:

    ... venv activate
    pip install rospkg defusedxml
    roslaunch foo bar.launch