orocos/orocos_kinematics_dynamics #275: Accuracy of KDL ChainIdSolver_RNE

DimitriEckert137, 2020-07-20:
Hi everyone,
I was wondering, have there been some tests of how accurate the results from the ChainIdSolver_RNE are? I tested with a simple inverted pendulum in gazebo and compared the results from kdl with the ones from gazebo and the theoretical value (calculated "by hand"). The results very pretty good, with kdl being slightly closer to the theoretical value. When I tested with a UR3e and compared kdl and gazebo (I didn't calculate the theoretical value here), the differences were much bigger. Has anybody else done some tests like this before and could tell me how accurate kdl is (under the idealistic assumption that the URDF model is perfectly accurate)?

YummyOu, 2021-05-14:
I have the same confusion.
I use UR5 to get the current from ROS and convert it to torque, but the differences are so big.

- A motionless posture
[image]
q:     3.140, -1.570, 1.570, -1.570, 4.775, 1.460
current:	-0.110, -1.791, -1.354, -0.223, 0.116, 0.067 	
measure_torque：  -1.389, -22.611, -17.094, -2.077, 1.080, 0.624
KDL_torque: 	       0.000, -11.050, -11.042, -1.306, 0.008, 0.000

- A movement
I send the velocity for joint1,2,5 to control the robot and the velocity curve like a trapezoidal wave.
torque image comparison:
[image]

So I also wonder how accurate the KDL for UR is.

luzpaz, 2023-09-24:
Any traction on this ?