orocos/orocos_kinematics_dynamics #277: PyKDL does not work after install ROS, error message:ImportError: /usr/local/lib/python3.6/dist-packages/PyKDL.so: undefined symbol: _ZN3KDL22ChainJntToJacDotSolver20E_FKSOLVERPOS_FAILEDE

MandyXie, 2020-07-30:
In Ubuntu 18.04, I correctly installed kdl and pykdl, and passed tests/PyKDLtest.py. However after I installed ROS, I cannot import PyKDL in my console anymore, it gives me the following error:
ImportError: /usr/local/lib/python3.6/dist-packages/PyKDL.so: undefined symbol: _ZN3KDL22ChainJntToJacDotSolver20E_FKSOLVERPOS_FAILEDE
/usr/local/lib/python3.6/dist-packages is where I installed pykdl.  I made sure that I sourced ROS first, then export PYTHONPATH=/usr/local/lib/python3.6/dist-packages/:$PYTHONPATH
If I just uninstall ROS, then everything works again.

MatthijsBurgh, 2020-07-30:
This has to be related to the pythonpath. So please paste the entire paths for both situations.

Please also provide the pythonpath in case that ROS is installed, but not sourced. Does it work in that case?

PS. Did you know you drop kdl and PyKDL in your catkin workspace and catkin will take care of it.

yoojinoh, 2020-08-25:
Hi, 

I receive the same error with PyKDL python3 with ROS Melodic.

If i don't source ROS related paths, then PyKDL seems to work when I import the library. But as soon as I add ROS paths, it links to PyKDL that's inside the ROS (which is for python2)

Is there a way that I can use both python3 PyKDL with ROS? I've tried to move around the order of imports but no progress..

MatthijsBurgh, 2020-08-25:
Please provide some information. Like PYTHONPATH and the paths of the libraries. Without any information I can't help.

PS make it a habit to include as much information as possible instead of a least as possible

yoojinoh, 2020-08-26:
My apologies and thank you in advance for your help. My goal is to use PyKDL along with ROS in a Python3 environment.

Here is my sys.path without sourcing ROS paths, and I can import PyKDL
['/home/venvs/tf2/bin', '/home/ros_ws/LeapSDK/lib', '/home/ros_ws/LeapSDK/lib/x64', '/home/venvs/tf2/lib/python36.zip', '/home/venvs/tf2/lib/python3.6', '/home/venvs/tf2/lib/python3.6/lib-dynload', '/usr/lib/python3.6', '', '/home/venvs/tf2/lib/python3.6/site-packages', '/home/venvs/tf2/lib/python3.6/site-packages/IPython/extensions', '/home/.ipython']

When I source the ROS workspace, it sems to read the PYKDL that's included in ROS (python2 version) when I would like to use PyKDL for python3
In this case I receive the error:
`ImportError: dynamic module does not define module export function (PyInit_PyKDL)
`
['/home/venvs/tf2/bin', '/home/projects/ros_ws/devel/lib/python2.7/dist-packages', '/home/projects/ros_ws/install/lib/python2.7/dist-packages', '/opt/ros/melodic/lib/python2.7/dist-packages', '/home/ros_ws/LeapSDK/lib', '/home/ros_ws/LeapSDK/lib/x64', '/home/venvs/tf2/lib/python36.zip', '/home/venvs/tf2/lib/python3.6', '/home/venvs/tf2/lib/python3.6/lib-dynload', '/usr/lib/python3.6', '/home/venvs/tf2/lib/python3.6/site-packages', '/home/venvs/tf2/lib/python3.6/site-packages/IPython/extensions', '/home/.ipython']

I tried to manually remove the ros path in Python using the following code
`sys.path.remove('/opt/ros/melodic/lib/python2.7/dist-packages')`

and then when I reimport PyKDL I got the error
`ImportError: /home/venvs/tf2/lib/python3.6/site-packages/PyKDL.so: undefined symbol: _ZN3KDL22ChainJntToJacDotSolver20E_FKSOLVERPOS_FAILEDE
`
The LD_LIBRARY_PATH includes the ros workspace lib directory path when I add source the ros devel/setup.bash

MatthijsBurgh, 2020-08-26:
@yoojinoh where is the PyKDL library located you want to use?

yoojinoh, 2020-08-26:
@MatthijsBurgh I would like to use this one
/home/projects/ros_ws/src/catkin/intent_prediction/orocos_kinematics_dynamics/python_orocos_kdl/build/PyKDL.so

Do you know how I could specify to do so?

MatthijsBurgh, 2020-08-26:
@yoojinoh set `ROS_PYTHON_VERSION` environment variable to 3 and then build with your catkin tool of preference. Then source your workspace.

MandyXie, 2020-08-26:
Thank you both! I was not able to follow-up my earlier post because I was using a laptop from my internship company, and I had to send the laptop back to the company before I got a chance to work on this issue. I just tried to install kdl in my personal machine, and I had the same issue again. I will try to follow your suggestion on setting ROS_PYTHON_VERSION.

MatthijsBurgh, 2020-10-25:
Closing, because of inactivity

closed 2020-10-25 by MatthijsBurgh