orocos/orocos_kinematics_dynamics #279: MacOS 10.15 Foxy python bindings build error

AndrewJSchoen, 2020-08-06:
I am attempting to build the PyKDL bindings on MacOS 10.15.6, python 3.8, and ROS2 Foxy. I cloned the most recent version and was able to build and install `orocos-kdl` without issue. To build PyKDL, I did the following:

```
cd python_orocos_kdl
mkdir build3 && cd build
export ROS_PYTHON_VERSION=3
cmake -DCMAKE_BUILD_TYPE=Release -DBUILD_PYKDL_PYBIND11=OFF ./..
make
```

At this step, I received the following error trace:
```
>> make
[ 10%] Generating PyKDL/sip/sipPyKDLpart0.cpp, PyKDL/sip/sipPyKDLpart1.cpp, PyKDL/sip/sipPyKDLpart2.cpp, PyKDL/sip/sipPyKDLpart3.cpp, PyKDL/sip/sipPyKDLpart4.cpp, PyKDL/sip/sipPyKDLpart5.cpp, PyKDL/sip/sipPyKDLpart6.cpp, PyKDL/sip/sipPyKDLpart7.cpp

Scanning dependencies of target python_module_PyKDL
[ 20%] Building CXX object CMakeFiles/python_module_PyKDL.dir/PyKDL/sip/sipPyKDLpart0.o
[ 30%] Building CXX object CMakeFiles/python_module_PyKDL.dir/PyKDL/sip/sipPyKDLpart1.o
[ 40%] Building CXX object CMakeFiles/python_module_PyKDL.dir/PyKDL/sip/sipPyKDLpart2.o
[ 50%] Building CXX object CMakeFiles/python_module_PyKDL.dir/PyKDL/sip/sipPyKDLpart3.o
[ 60%] Building CXX object CMakeFiles/python_module_PyKDL.dir/PyKDL/sip/sipPyKDLpart4.o
/path/to/orocos_kinematics_dynamics/python_orocos_kdl/build3/PyKDL/sip/sipPyKDLpart4.cpp:1423:46: error: no member named 'Fixed' in 'KDL::Joint'
        const  ::Joint& a1def = Joint(Joint::Fixed);
                                      ~~~~~~~^
/path/to/orocos_kinematics_dynamics/python_orocos_kdl/build3/PyKDL/sip/sipPyKDLpart4.cpp:1447:46: error: no member named 'Fixed' in 'KDL::Joint'
        const  ::Joint& a0def = Joint(Joint::Fixed);
                                      ~~~~~~~^
2 errors generated.
make[2]: *** [CMakeFiles/python_module_PyKDL.dir/PyKDL/sip/sipPyKDLpart4.o] Error 1
make[1]: *** [CMakeFiles/python_module_PyKDL.dir/all] Error 2
make: *** [all] Error 2
```

When building with PyBind (`cmake -DCMAKE_BUILD_TYPE=Release -DBUILD_PYKDL_PYBIND11=OFF ./..`), I get a different error:
```
>> make
Scanning dependencies of target PyKDL
[ 16%] Building CXX object CMakeFiles/PyKDL.dir/PyKDL/pybind11/PyKDL.o
[ 33%] Building CXX object CMakeFiles/PyKDL.dir/PyKDL/pybind11/frames.o
In file included from /path/to/orocos_kinematics_dynamics/python_orocos_kdl/PyKDL/pybind11/frames.cpp:27:
In file included from /path/to/orocos_kinematics_dynamics/python_orocos_kdl/PyKDL/pybind11/PyKDL.h:25:
/path/to/orocos_kinematics_dynamics/python_orocos_kdl/pybind11/include/pybind11/pybind11.h:133:9: error: static_assert failed due to requirement 'expected_num_args<pybind11::name,
      pybind11::is_method, pybind11::sibling, pybind11::arg_v>(sizeof...(Args), argument_loader<const KDL::Vector *>::has_args, argument_loader<const KDL::Vector *>::has_kwargs)' "The number of argument
      annotations does not match the number of function arguments"
        static_assert(expected_num_args<Extra...>(sizeof...(Args), cast_in::has_args, cast_in::has_kwargs),
        ^             ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/path/to/orocos_kinematics_dynamics/python_orocos_kdl/pybind11/include/pybind11/pybind11.h:85:9: note: in instantiation of function template specialization
      'pybind11::cpp_function::initialize<(lambda at /path/to/orocos_kinematics_dynamics/python_orocos_kdl/pybind11/include/pybind11/pybind11.h:85:20), double, const KDL::Vector *,
      pybind11::name, pybind11::is_method, pybind11::sibling, pybind11::arg_v>' requested here
        initialize([f](const Class *c, Arg... args) -> Return { return (c->*f)(args...); },
        ^
/path/to/orocos_kinematics_dynamics/python_orocos_kdl/pybind11/include/pybind11/pybind11.h:1118:22: note: in instantiation of function template specialization
      'pybind11::cpp_function::cpp_function<double, KDL::Vector, pybind11::name, pybind11::is_method, pybind11::sibling, pybind11::arg_v>' requested here
        cpp_function cf(method_adaptor<type>(std::forward<Func>(f)), name(name_), is_method(*this),
                     ^
/path/to/orocos_kinematics_dynamics/python_orocos_kdl/PyKDL/pybind11/frames.cpp:92:12: note: in instantiation of function template specialization
      'pybind11::class_<KDL::Vector>::def<double (KDL::Vector::*)() const, pybind11::arg_v>' requested here
    vector.def("Norm", &Vector::Norm, py::arg("eps")=epsilon);
           ^
1 error generated.
make[2]: *** [CMakeFiles/PyKDL.dir/PyKDL/pybind11/frames.o] Error 1
make[1]: *** [CMakeFiles/PyKDL.dir/all] Error 2
make: *** [all] Error 2
```

Any help/suggestions would be appreciated!

MatthijsBurgh, 2020-08-07:
@meyerj has been working on adding MacOS support. So he is probably to guy to help you.
(I don't think we already fixed the issues)

AndrewJSchoen, 2020-08-10:
Thanks for the heads up. @meyerj any suggestions on how to proceed?

AndrewJSchoen, 2020-08-10:
Actually, I just tried building with PyBind11, on the following branch: https://github.com/orocos/orocos_kinematics_dynamics/tree/refactor/travis-include-osx
And successfully built PyKDL. Building with SIP is still broken, though.

MatthijsBurgh, 2020-08-12:
@AndrewJSchoen could you merge master into that branch and test if that also works?

AndrewJSchoen, 2020-08-12:
It looks like rebuilding the PyKDL with the merged refactor/master produces the following error:

```
In file included from /path/to/orocos_kinematics_dynamics/python_orocos_kdl/PyKDL/pybind11/frames.cpp:27:
In file included from /path/to/orocos_kinematics_dynamics/python_orocos_kdl/PyKDL/pybind11/PyKDL.h:25:
/path/to/orocos_kinematics_dynamics/python_orocos_kdl/pybind11/include/pybind11/pybind11.h:133:9: error: static_assert failed due to requirement 'expected_num_args<pybind11::name, pybind11::is_method, pybind11::sibling,
      pybind11::arg_v>(sizeof...(Args), argument_loader<const KDL::Vector *>::has_args, argument_loader<const KDL::Vector *>::has_kwargs)' "The number of argument annotations does not match the number of function arguments"
        static_assert(expected_num_args<Extra...>(sizeof...(Args), cast_in::has_args, cast_in::has_kwargs),
        ^             ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/path/to/orocos_kinematics_dynamics/python_orocos_kdl/pybind11/include/pybind11/pybind11.h:85:9: note: in instantiation of function template specialization 'pybind11::cpp_function::initialize<(lambda at
      /path/to/orocos_kinematics_dynamics/python_orocos_kdl/pybind11/include/pybind11/pybind11.h:85:20), double, const KDL::Vector *, pybind11::name, pybind11::is_method, pybind11::sibling, pybind11::arg_v>' requested here
        initialize([f](const Class *c, Arg... args) -> Return { return (c->*f)(args...); },
        ^
/path/to/orocos_kinematics_dynamics/python_orocos_kdl/pybind11/include/pybind11/pybind11.h:1118:22: note: in instantiation of function template specialization 'pybind11::cpp_function::cpp_function<double, KDL::Vector, pybind11::name,
      pybind11::is_method, pybind11::sibling, pybind11::arg_v>' requested here
        cpp_function cf(method_adaptor<type>(std::forward<Func>(f)), name(name_), is_method(*this),
                     ^
/path/to/orocos_kinematics_dynamics/python_orocos_kdl/PyKDL/pybind11/frames.cpp:92:12: note: in instantiation of function template specialization 'pybind11::class_<KDL::Vector>::def<double (KDL::Vector::*)() const, pybind11::arg_v>'
      requested here
    vector.def("Norm", &Vector::Norm, py::arg("eps")=epsilon);
           ^
1 error generated.
make[2]: *** [CMakeFiles/PyKDL.dir/PyKDL/pybind11/frames.o] Error 1
make[1]: *** [CMakeFiles/PyKDL.dir/all] Error 2
make: *** [all] Error 2
```

MatthijsBurgh, 2020-08-12:
That is so weird as the PyBind files are not changed. Also the PyBind submodule is that same. Also the changes made in the cpp library are not in `Vector::Norm` or any related file.

MatthijsBurgh, 2020-08-21:
@AndrewJSchoen Could you maybe figure out, which commit on master breaks the `refactor/travis-include-osx` branch. Hard to debug with only CI.

MatthijsBurgh, 2020-09-15:
@AndrewJSchoen Have you already been able to pinpoint the specific commit?

meyerj, 2020-09-24:
This problem sounds similar as https://github.com/orocos/orocos_kinematics_dynamics/issues/287 and might not be related to macOS at all. Which version of `orocos_kdl` you build against? `python_orocos_kdl` is currently coupled to the version of `orocos_kdl` in the same commit, and you cannot build `python_orocos_kdl` for an older version of `orocos_kdl` installed elsewhere, e.g. through [HomeBrew](https://formulae.brew.sh/formula/orocos-kdl).

If that is indeed the case, see https://github.com/orocos/orocos_kinematics_dynamics/issues/287#issuecomment-698229866 for a list of potential solutions.

AndrewJSchoen, 2020-09-24:
Sorry about the delay. Been super busy, and since I somehow got it working, I haven't wanted to fiddle with it much. My guess is that as the linked issue mentions, it was grabbing the wrong version of the orocos library, causing the build to fail. I say this because when I isolated the setup from anything else, I got inconsistent results whether I was in a ROS environment. I was helping someone else out on ubuntu, and they had the same issues, so it likely isn't a Mac-specific error.

closed 2021-02-08 by MatthijsBurgh