orocos/orocos_kinematics_dynamics #283: Building with `colcon --symlink-install` succeeds, but does add to correct PYTHONPATH after sourcing setup.bash

daenny, 2020-08-20:
I tried buidling this packages with colcon. The package build succeeds, but afterwards it cannot be imported.

I ran:
`colcon build --catkin-skip-building-tests --symlink-install --build-base ../build_colcon --install-base ../devel`
Which succeeds. The `PyKDL.so` is also correctly build and located in the devel folder.

But when I want to use it:
Traceback:
```
-- stderr: sr_ade                                                                                       
Traceback (most recent call last):
import tf_conversions.posemath as tfkdlc
  File "/home/daniel/ros/melodic/smart_robotics/devel/tf_conversions/lib/python2.7/dist-packages/tf_conversions/__init__.py", line 30, in <module>
    from .posemath import *
  File "/home/daniel/ros/melodic/smart_robotics/devel/tf_conversions/lib/python2.7/dist-packages/tf_conversions/posemath.py", line 34, in <module>
    from PyKDL import *
ImportError: No module named PyKDL
```

for some reason the `python_orocos_kdl` folder is not added to the `PYTHONPATH` and also not to the `ROS_PACKAGE_PATH`, while other ROS packages are. In the sourced packages `share` folder is a subfolder called `hook` which does not contain the source information in `python_orocos_kdl`.

`python_orocos_kdl`:

```
daniel@daniel-dell:~/ros/melodic/smart_robotics/devel/python_orocos_kdl/share/python_orocos_kdl $ ag PYTH
catkin_env_hook/python_orocos_kdl_site_packages.tcsh
1:export PYTHONPATH=/home/daniel/ros/melodic/smart_robotics/devel/python_orocos_kdl/lib/python2.7/dist-packages:$PYTHONPATH

catkin_env_hook/python_orocos_kdl_site_packages.zsh
1:export PYTHONPATH=/home/daniel/ros/melodic/smart_robotics/devel/python_orocos_kdl/lib/python2.7/dist-packages:$PYTHONPATH

catkin_env_hook/python_orocos_kdl_site_packages.bash
1:export PYTHONPATH=/home/daniel/ros/melodic/smart_robotics/devel/python_orocos_kdl/lib/python2.7/dist-packages:$PYTHONPATH
```

vs

`sr_world_model`:

```
daniel@daniel-dell:~/ros/.../share/sr_world_model $ ag PYTH
hook/python2path.dsv
1:prepend-non-duplicate;PYTHONPATH;lib/python2.7/dist-packages

hook/python2path.ps1
3:colcon_prepend_unique_value PYTHONPATH "$env:COLCON_CURRENT_PREFIX\lib/python2.7/dist-packages"

hook/python2path.sh
3:_colcon_prepend_unique_value PYTHONPATH "$COLCON_CURRENT_PREFIX/lib/python2.7/dist-packages"

hook/catkin_pythonpath.ps1
3:colcon_prepend_unique_value PYTHONPATH "$env:COLCON_CURRENT_PREFIX\lib/python2.7/dist-packages"

hook/catkin_pythonpath.dsv
1:prepend-non-duplicate;PYTHONPATH;lib/python2.7/dist-packages

hook/catkin_pythonpath.sh
3:_colcon_prepend_unique_value PYTHONPATH "$COLCON_CURRENT_PREFIX/lib/python2.7/dist-packages"
```

meyerj, 2020-08-31:
The problem is that `python_orocos_kdl` (and also `orocos_kdl`) is declared with `<build_type>cmake</build_type>` in [package.xml](https://github.com/orocos/orocos_kinematics_dynamics/blob/29f2329eb8e2f9b96d0c48c5cd21c342d420d212/python_orocos_kdl/package.xml#L35). So colcon treats it like a "normal" cmake package and does not invoke the special extensions for catkin from [colcon_ros/task/catkin/build.py](https://github.com/colcon/colcon-ros/blob/d115601d534b0ccf5644ee58450ececcab505933/colcon_ros/task/catkin/build.py).

`catkin` is still found and used conditionally in
https://github.com/orocos/orocos_kinematics_dynamics/blob/29f2329eb8e2f9b96d0c48c5cd21c342d420d212/python_orocos_kdl/CMakeLists.txt#L59-L62

just for the sake of installing an env-hook, but neither of the two packages is actually a real catkin package and supports being built in a pure devel-space.

I see two possible solutions, with variants:

1. a. Remove `<build_type>cmake</build_type>` from `package.xml`. This would break building the package without installation with [catkin_tools](https://catkin-tools.readthedocs.io/en/latest/) or `catkin_make_isolated`, which "install" cmake packages to the devel-space in this case. For users that do not build with any of the ROS tools it has no effect, because `package.xml` is not needed anyway.

   b. Same as a., but with full catkin support, including devel-space builds: This would require quite significant changes, as the structure and names of header files in the package source must match the ones from the install-space, which is currently not the case for `orocos_kdl`. `python_orocos_kdl` has no headers, so probably it is easier to apply there. To still support non-ROS builds, the `CMakeLists.txt` would require a number of conditional sections. _Probably not the way to go_...

2. Add a [`colcon.pkg` file](https://colcon.readthedocs.io/en/released/user/configuration.html#colcon-pkg-files) that explicitly lists the env-hook installed to `install/share/python_orocos_kdl/catkin_env_hook/python_orocos_kdl_site_packages.sh`.

I personally would prefer the latter, and we already applied that for [orocos_toolchain](https://github.com/orocos-toolchain/orocos_toolchain) in the [ros2](https://github.com/orocos-toolchain/orocos_toolchain/blob/ros2/colcon.pkg) branch (PR is pending).

Note that it is not necessary to provide [multiple, shell-specific env-hooks](https://github.com/orocos/orocos_kinematics_dynamics/tree/29f2329eb8e2f9b96d0c48c5cd21c342d420d212/python_orocos_kdl/env-hooks), neither for catkin nor for colcon. A single `.sh` hook is sufficient, and also sourced by `bash` and `zsh`, if no more specific env-hook exists.

On my system (Ubuntu 18.04), `PYTHON_SITE_PACKAGES_INSTALL_DIR` is set to `<prefix>/lib/python2.7/dist-packages`. If that is the case, the extra env-hook is not even needed, because catkin (or colcon with the catkin extension) adds this path to the `PYTHONPATH` internally already unconditionally. Apparently the installation path was changed in https://github.com/orocos/orocos_kinematics_dynamics/commit/4634955afc93871d32e55bb9903f160bf5a46847. But note that `dist-packages` is [Debian/Ubuntu-specific](https://stackoverflow.com/questions/9387928/whats-the-difference-between-dist-packages-and-site-packages) - not sure what side effects it may have to install `PyKDL.so` to `dist-packages` on macOS, for example? With catkin, `PYTHON_SITE_PACKAGES_INSTALL_DIR` could simply be set to `CATKIN_GLOBAL_PYTHON_DESTINATION` without any env-hook, which is set to `PYTHON_INSTALL_DIR` configured [here](https://github.com/ros/catkin/blob/dfc93c72d9799f57442d08c2f6e9fd494bba95c1/cmake/python.cmake). Of course not providing the env-hook at all would also not work for colcon, without any of the other solutions above. And for non-ROS builds, if catkin is not available, `PYTHON_SITE_PACKAGES_INSTALL_DIR` still has to be set to a reasonable, system-specific default value...

meyerj, 2020-08-31:
I proposed a patch in #285, which implements 1a. and partially 1b. (for `python_orocos_kdl`). Not requiring env-hooks at all and relying on catkin and colcon-ros to setup the `PYTHONPATH` is probably the easiest and least error-prone solution. Non-ROS users probably install to a default location like `/usr` or `/usr/local`, where the env-hook is not needed and was also not installed before.

daenny, 2020-08-31:
Thank you both for testing and the elaborate explanations. I might have some time for testing on my machine tomorrow, but I suspect that your solution works for our use case.

closed 2020-09-21 by MatthijsBurgh