orocos/orocos_kinematics_dynamics #284: undefined symbol: _ZNK3KDL6Vector4NormEd

libing64, 2020-08-29:
./examples/geometry 
./examples/geometry: symbol lookup error: ./examples/geometry: undefined symbol: _ZNK3KDL6Vector4NormEd

=============================
trajectory_example and chainiksolverpos_lma_demo works well, but geometry has error when runing.

libing64, 2020-08-29:
//Calculating the norm and normalising your vector
    std::cout<<"norm(v3): "<<v3.Norm()<<std::endl;
=================================
The error seems caused by this line, after commenting this line, geometry example works well.

MatthijsBurgh, 2020-08-29:
Thanks for the heads up. I hope it doesn’t block you for now.

MatthijsBurgh, 2020-08-29:
I have tried it on my own system, but it works fine. So please provide more information about your system and procedures

libing64, 2020-09-05:
@MatthijsBurgh Ubuntu18.04.5 LTS

boost 1.65
Eigen 3.3.4-4

MatthijsBurgh, 2020-09-12:
I have implemented it on travis, see https://travis-ci.org/github/orocos/orocos_kinematics_dynamics/builds/726663753. It works fine, similar to my system, which also runs 18.04. so just try removing all build files and build again.

libing64, 2020-09-13:
@MatthijsBurgh After removing all build files and rebuild the code, geomery example works fine. 
Please close this issue and great thanks for your reply.

closed 2020-09-13 by libing64