orocos/orocos_kinematics_dynamics #287: cannot compile and build orocos_kinematic_dynamics. PLEASE HELP!

radhen, 2020-09-23:
I am trying to build the master branch of this repository with python3.6. I am following the steps mentioned [here](https://github.com/orocos/orocos_kinematics_dynamics/issues/115). I am also doing `git submodule init && git submodule update` to get the pybind11 files. I can successfully build the orocos_kdl folder but not the python_orocos_kdl folder. I keep getting errors which are hard to find on google. I am pasting just a few lines of errors here for your reference. 

--------------------------------------------------------------- ERROR I GET ------------------------------------------------------------------------------------
`/home/radhen/Documents/orocos_kinematics_dynamics/python_orocos_kdl/PyKDL/pybind11/kinfam.cpp: In function ‘void init_kinfam(pybind11::module&)’:
/home/radhen/Documents/orocos_kinematics_dynamics/python_orocos_kdl/PyKDL/pybind11/kinfam.cpp:67:53: error: ‘Fixed’ is not a member of ‘KDL::Joint::JointType’
         joint_type.value("Fixed", Joint::JointType::Fixed);
                                                     ^~~~~
/home/radhen/Documents/orocos_kinematics_dynamics/python_orocos_kdl/PyKDL/pybind11/kinfam.cpp:75:66: error: ‘Fixed’ is not a member of ‘KDL::Joint::JointType’
               py::arg("name"), py::arg("type")=Joint::JointType::Fixed, py::arg("scale")=1, py::arg("offset")=0,
                                                                  ^~~~~
/home/radhen/Documents/orocos_kinematics_dynamics/python_orocos_kdl/PyKDL/pybind11/kinfam.cpp:78:49: error: ‘Fixed’ is not a member of ‘KDL::Joint::JointType’
               py::arg("type")=Joint::JointType::Fixed, py::arg("scale")=1, py::arg("offset")=0,
                                                 ^~~~~
/home/radhen/Documents/orocos_kinematics_dynamics/python_orocos_kdl/PyKDL/pybind11/kinfam.cpp:493:90: error: ‘INERTIAL’ is not a member of ‘KDL::ChainJntToJacDotSolver’
     chain_jnt_to_jac_dot_solver.def_readonly_static("INERTIAL", &ChainJntToJacDotSolver::INERTIAL);
                                                                                          ^~~~~~~~
/home/radhen/Documents/orocos_kinematics_dynamics/python_orocos_kdl/PyKDL/pybind11/kinfam.cpp:497:91: error: ‘setInertialRepresentation’ is not a member of ‘KDL::ChainJntToJacDotSolver’
     chain_jnt_to_jac_dot_solver.def("setInertialRepresentation", &ChainJntToJacDotSolver::setInertialRepresentation);
                                                                                           ^~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/radhen/Documents/orocos_kinematics_dynamics/python_orocos_kdl/PyKDL/pybind11/PyKDL.h:27:0,
                 from /home/radhen/Documents/orocos_kinematics_dynamics/python_orocos_kdl/PyKDL/pybind11/framevel.cpp:27:
/home/radhen/Documents/orocos_kinematics_dynamics/python_orocos_kdl/pybind11/include/pybind11/operators.h: In instantiation of ‘struct pybind11::detail::op_impl<(pybind11::detail::op_id)25, (pybind11::detail::op_type)0, KDL::Rall1d<double>, KDL::Rall1d<double>, KDL::Rall1d<double> >’:
/home/radhen/Documents/orocos_kinematics_dynamics/python_orocos_kdl/pybind11/include/pybind11/operators.h:59:24:   required from ‘void pybind11::detail::op_<id, ot, L, R>::execute(Class&, const Extra& ...) const [with Class = pybind11::class_<KDL::Rall1d<double> >; Extra = {}; pybind11::detail::op_id id = (pybind11::detail::op_id)25; pybind11::detail::op_type ot = (pybind11::detail::op_type)0; L = pybind11::detail::self_t; R = pybind11::detail::self_t]’
/home/radhen/Documents/orocos_kinematics_dynamics/python_orocos_kdl/pybind11/include/pybind11/pybind11.h:1136:9:   required from ‘pybind11::class_<type_, options>& pybind11::class_<type_, options>::def(const pybind11::detail::op_<id, ot, L, R>&, const Extra& ...) [with pybind11::detail::op_id id = (pybind11::detail::op_id)25; pybind11::detail::op_type ot = (pybind11::detail::op_type)0; L = pybind11::detail::self_t; R = pybind11::detail::self_t; Extra = {}; type_ = KDL::Rall1d<double>; options = {}]’
/home/radhen/Documents/orocos_kinematics_dynamics/python_orocos_kdl/PyKDL/pybind11/framevel.cpp:62:40:   required from here
/home/radhen/Documents/orocos_kinematics_dynamics/python_orocos_kdl/pybind11/include/pybind11/operators.h:130:67: error: no match for ‘operator==’ (operand types are ‘const KDL::Rall1d<double>’ and ‘const KDL::Rall1d<double>’)
 PYBIND11_BINARY_OPERATOR(eq,        eq,           operator==,   l == r)
`
--------------------------------------------------------------- ERROR I GET ------------------------------------------------------------------------------------

I believe there is some issue with the python version. I have ROS melodic installed and also python_orocos_kdl installed in ROS which uses python2. I have tried removing python_orocos_kdl from ROS and build this git package but still no luck. I am not exactly sure how to go about debugging this. Please help me resolve this issue. Thank you for reading :)

MatthijsBurgh, 2020-09-23:
It is pretty clear, it can't find some of the member functions/variables. Please start completely over and follow the build instructions from the travis config, https://github.com/orocos/orocos_kinematics_dynamics/blob/master/.travis.yml

MatthijsBurgh, 2020-09-23:
How about building the python2 version?

radhen, 2020-09-23:
so i tried building the repo using sip instead of pybind11 by setting these flags, 
` OROCOS_KDL_BUILD_TYPE=Release BUILD_PYKDL_PYBIND11=OFF`

still can't build python_orocos_kdl folder. I get this error, 

 ------------------------------------------------------------- ERROR I GET (complete) -----------------------------------------------------------------------
/home/radhen/Documents/orocos_kinematics_dynamics/python_orocos_kdl/build3/PyKDL/sip/sipPyKDLpart1.cpp: In function ‘PyObject* meth_ChainJntToJacDotSolver_setInertialRepresentation(PyObject*, PyObject*)’:
/home/radhen/Documents/orocos_kinematics_dynamics/python_orocos_kdl/build3/PyKDL/sip/sipPyKDLpart1.cpp:281:21: error: ‘class KDL::ChainJntToJacDotSolver’ has no member named ‘setInertialRepresentation’; did you mean ‘setInternialRepresentation’?
             sipCpp->setInertialRepresentation();
                     ^~~~~~~~~~~~~~~~~~~~~~~~~
                     setInternialRepresentation
/home/radhen/Documents/orocos_kinematics_dynamics/python_orocos_kdl/build3/PyKDL/sip/sipPyKDLpart1.cpp: In function ‘PyObject* varget_ChainJntToJacDotSolver_INERTIAL(void*, PyObject*, PyObject*)’:
/home/radhen/Documents/orocos_kinematics_dynamics/python_orocos_kdl/build3/PyKDL/sip/sipPyKDLpart1.cpp:477:41: error: ‘INERTIAL’ is not a member of ‘KDL::ChainJntToJacDotSolver’
     sipVal =  ::ChainJntToJacDotSolver::INERTIAL;
                                         ^~~~~~~~
CMakeFiles/python_module_PyKDL.dir/build.make:119: recipe for target 'CMakeFiles/python_module_PyKDL.dir/PyKDL/sip/sipPyKDLpart1.cpp.o' failed
make[2]: *** [CMakeFiles/python_module_PyKDL.dir/PyKDL/sip/sipPyKDLpart1.cpp.o] Error 1
CMakeFiles/Makefile2:227: recipe for target 'CMakeFiles/python_module_PyKDL.dir/all' failed
make[1]: *** [CMakeFiles/python_module_PyKDL.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
 ------------------------------------------------------------- ERROR I GET (complete) -----------------------------------------------------------------------

As you suggested I also tried using python2. I set the flags
` OROCOS_KDL_BUILD_TYPE=Release BUILD_PYKDL_PYBIND11=ON`

and also change this command,
`export ROS_PYTHON_VERSION=2`

 ------------------------------------------------------------- ERROR I GET (complete) -----------------------------------------------------------------------
`In file included from /home/radhen/Documents/orocos_kinematics_dynamics/python_orocos_kdl/PyKDL/pybind11/PyKDL.h:25:0,
                 from /home/radhen/Documents/orocos_kinematics_dynamics/python_orocos_kdl/PyKDL/pybind11/frames.cpp:27:
/home/radhen/Documents/orocos_kinematics_dynamics/python_orocos_kdl/pybind11/include/pybind11/pybind11.h: In instantiation of ‘void pybind11::cpp_function::initialize(Func&&, Return (*)(Args ...), const Extra& ...) [with Func = pybind11::cpp_function::cpp_function(Return (Class::*)(Arg ...) const, const Extra& ...) [with Return = double; Class = KDL::Vector; Arg = {}; Extra = {pybind11::name, pybind11::is_method, pybind11::sibling, pybind11::arg_v}]::<lambda(const KDL::Vector*)>; Return = double; Args = {const KDL::Vector*}; Extra = {pybind11::name, pybind11::is_method, pybind11::sibling, pybind11::arg_v}]’:
/home/radhen/Documents/orocos_kinematics_dynamics/python_orocos_kdl/pybind11/include/pybind11/pybind11.h:85:9:   required from ‘pybind11::cpp_function::cpp_function(Return (Class::*)(Arg ...) const, const Extra& ...) [with Return = double; Class = KDL::Vector; Arg = {}; Extra = {pybind11::name, pybind11::is_method, pybind11::sibling, pybind11::arg_v}]’
/home/radhen/Documents/orocos_kinematics_dynamics/python_orocos_kdl/pybind11/include/pybind11/pybind11.h:1118:22:   required from ‘pybind11::class_<type_, options>& pybind11::class_<type_, options>::def(const char*, Func&&, const Extra& ...) [with Func = double (KDL::Vector::*)() const; Extra = {pybind11::arg_v}; type_ = KDL::Vector; options = {}]’
/home/radhen/Documents/orocos_kinematics_dynamics/python_orocos_kdl/PyKDL/pybind11/frames.cpp:92:61:   required from here
/home/radhen/Documents/orocos_kinematics_dynamics/python_orocos_kdl/pybind11/include/pybind11/pybind11.h:133:9: error: static assertion failed: The number of argument annotations does not match the number of function arguments
         static_assert(expected_num_args<Extra...>(sizeof...(Args), cast_in::has_args, cast_in::has_kwargs),
         ^~~~~~~~~~~~~
/home/radhen/Documents/orocos_kinematics_dynamics/python_orocos_kdl/pybind11/include/pybind11/pybind11.h:137:19: error: cannot convert ‘pybind11::cpp_function::initialize(Func&&, Return (*)(Args ...), const Extra& ...) [with Func = pybind11::cpp_function::cpp_function(Return (Class::*)(Arg ...) const, const Extra& ...) [with Return = double; Class = KDL::Vector; Arg = {}; Extra = {pybind11::name, pybind11::is_method, pybind11::sibling, pybind11::arg_v}]::<lambda(const KDL::Vector*)>; Return = double; Args = {const KDL::Vector*}; Extra = {pybind11::name, pybind11::is_method, pybind11::sibling, pybind11::arg_v}]::<lambda(pybind11::detail::function_call&)>’ to ‘pybind11::handle (*)(pybind11::detail::function_call&)’ in assignment
         rec->impl = [](function_call &call) -> handle {
CMakeFiles/PyKDL.dir/build.make:86: recipe for target 'CMakeFiles/PyKDL.dir/PyKDL/pybind11/frames.cpp.o' failed
make[2]: *** [CMakeFiles/PyKDL.dir/PyKDL/pybind11/frames.cpp.o] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/PyKDL.dir/all' failed
[... cut ...]
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2`
 ------------------------------------------------------------- ERROR I GET (complete) -----------------------------------------------------------------------

seems like its the same error to me. 

I can install PyKDL in python2 using ROS by executing this command, 
`sudo apt-get install ros-melodic-python-orocos-kdl`
and things work. I have been using PyKDL in python2 in the past. 

My new files are in python3 env. and so I am desperate to install PyKDL in python3.

MatthijsBurgh, 2020-09-23:
It seems to me that `orocos_kdl` isn't building correct and therefore the linking doesn't work. I have no clue what causes the problems with the cpp library.

radhen, 2020-09-23:
how come we are building the folder orocos_kdl without specifying the python version (default python2) while for the python_orocos_kdl folder we specific which python verison we want?

MatthijsBurgh, 2020-09-23:
`orocos_kdl` is a cpp library....

radhen, 2020-09-23:
makes sense. sorry for the dumb question. as far as i can tell, the orocos_kdl library compiles well. No errors. is there any other debugging step you can suggest?

radhen, 2020-09-23:
I tried the steps suggested (https://github.com/orocos/orocos_kinematics_dynamics/blob/master/.travis.yml) to me initially on a fresh PC. This fresh PC has python 3.6.9 and ROS melodic. I am still getting the same errors. Nothing changed. Can anyone please suggest me any steps to debug/solve this??

Below is full error again for your reference

`In file included from /home/gelsight/Documents/orocos_kinematics_dynamics/python_orocos_kdl/PyKDL/pybind11/PyKDL.h:25:0,
                 from /home/gelsight/Documents/orocos_kinematics_dynamics/python_orocos_kdl/PyKDL/pybind11/frames.cpp:27:
/home/gelsight/Documents/orocos_kinematics_dynamics/python_orocos_kdl/pybind11/include/pybind11/pybind11.h: In instantiation of ‘void pybind11::cpp_function::initialize(Func&&, Return (*)(Args ...), const Extra& ...) [with Func = pybind11::cpp_function::cpp_function(Return (Class::*)(Arg ...) const, const Extra& ...) [with Return = double; Class = KDL::Vector; Arg = {}; Extra = {pybind11::name, pybind11::is_method, pybind11::sibling, pybind11::arg_v}]::<lambda(const KDL::Vector*)>; Return = double; Args = {const KDL::Vector*}; Extra = {pybind11::name, pybind11::is_method, pybind11::sibling, pybind11::arg_v}]’:
/home/gelsight/Documents/orocos_kinematics_dynamics/python_orocos_kdl/pybind11/include/pybind11/pybind11.h:85:9:   required from ‘pybind11::cpp_function::cpp_function(Return (Class::*)(Arg ...) const, const Extra& ...) [with Return = double; Class = KDL::Vector; Arg = {}; Extra = {pybind11::name, pybind11::is_method, pybind11::sibling, pybind11::arg_v}]’
/home/gelsight/Documents/orocos_kinematics_dynamics/python_orocos_kdl/pybind11/include/pybind11/pybind11.h:1118:22:   required from ‘pybind11::class_<type_, options>& pybind11::class_<type_, options>::def(const char*, Func&&, const Extra& ...) [with Func = double (KDL::Vector::*)() const; Extra = {pybind11::arg_v}; type_ = KDL::Vector; options = {}]’
/home/gelsight/Documents/orocos_kinematics_dynamics/python_orocos_kdl/PyKDL/pybind11/frames.cpp:92:61:   required from here
/home/gelsight/Documents/orocos_kinematics_dynamics/python_orocos_kdl/pybind11/include/pybind11/pybind11.h:133:9: error: static assertion failed: The number of argument annotations does not match the number of function arguments
         static_assert(expected_num_args<Extra...>(sizeof...(Args), cast_in::has_args, cast_in::has_kwargs),
         ^~~~~~~~~~~~~
/home/gelsight/Documents/orocos_kinematics_dynamics/python_orocos_kdl/pybind11/include/pybind11/pybind11.h:137:19: error: cannot convert ‘pybind11::cpp_function::initialize(Func&&, Return (*)(Args ...), const Extra& ...) [with Func = pybind11::cpp_function::cpp_function(Return (Class::*)(Arg ...) const, const Extra& ...) [with Return = double; Class = KDL::Vector; Arg = {}; Extra = {pybind11::name, pybind11::is_method, pybind11::sibling, pybind11::arg_v}]::<lambda(const KDL::Vector*)>; Return = double; Args = {const KDL::Vector*}; Extra = {pybind11::name, pybind11::is_method, pybind11::sibling, pybind11::arg_v}]::<lambda(pybind11::detail::function_call&)>’ to ‘pybind11::handle (*)(pybind11::detail::function_call&)’ in assignment
         rec->impl = [](function_call &call) -> handle {
CMakeFiles/PyKDL.dir/build.make:86: recipe for target 'CMakeFiles/PyKDL.dir/PyKDL/pybind11/frames.cpp.o' failed
make[2]: *** [CMakeFiles/PyKDL.dir/PyKDL/pybind11/frames.cpp.o] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/PyKDL.dir/all' failed
make[1]: *** [CMakeFiles/PyKDL.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2`

meyerj, 2020-09-24:
I did not look at it in detail not test it myself, but the problem could be that you are mixing different versions of KDL. You cannot build and install `orocos_kdl` from source if you also have another version installed from debians in `/opt/ros` or even from a system package in `/usr`. Likely they will conflict, and `python_orocos_kdl` finds the library and headers of the ROS version at `/opt/ros` instead of the one you installed to `/usr/local`. It may be an option to uninstall `ros-melodic-orocos-kdl` completely, but then you also have to build all other ROS packages that depend on it from source.

And that is the root cause of the problem then: The latest version of KDL that is released into ROS, [1.4.0](https://github.com/orocos/orocos_kinematics_dynamics/compare/v1.4.0...master), is not API-compatible to the version of `python_orocos_kdl` in the `master` branch, and the version in 1.4.0 does not support Python 3 yet. So in short: It is not possible to build `python_orocos_kdl` for Python 3 in a ROS environment at the moment without hassles.

#### Possible solutions:

1. Make `python_orocos_kdl` in `master` compatible to the latest released version 1.4, using cmake magic and/or preprocessor conditionals or whatever is the best way to do this with sip and/or pybind11. Probably that would be hard to maintain if all future updates of `python_orocos_kdl` must be backwards-compatible to one or more released versions of `orocos_kdl`, and not only to the version in the exact same commit...

2. Backport the new `python_orocos_kdl` package and eventually other newer patches to an ABI-compatible 1.4 release branch (i.e. `release-1.4`) and draft a new release of 1.4.1. Then you could use that branch to build `python_orocos_kdl` from source for Python 3, against the released version 1.4.0 or 1.4.1 of `orocos_kdl`.

3. Build orocos_kdl as a static library with position-independent code, and then make sure to link the static archive into the Python module built as a shared library in `python_orocos_kdl` (i.e. by not installing `orocos_kdl` to a default location and not sourcing a ROS workspace before building), such that it does not link to `liborocos_kdl.so` provided elsewhere or find other installed headers anymore. Not sure whether the current CMake scripts would support that approach out-of-the-box, by setting the respective CMake options.

@MatthijsBurgh @smits Any other thoughts on this? Would 1. be feasible? Or does someone have time to work towards a new backwards-compatible 1.4 release? That would probably be the best and least error-prone solution.

MatthijsBurgh, 2020-09-24:
I think it a good thing to create a new release 1.5 for noetic etc.

I don't think we should provide backward compatibility. So an 1.4.1 release sound like an option. I don't have the knowledge about what is ABI compatible and what not. Also I don't have the time to do this by myself, if I had that knowledge.

If we go for that option you or @smits would need to determine what is ABI compatible and then we could split the tasks.

radhen, 2020-09-24:
Thank you @MatthijsBurgh and @meyerj for your comments. I am certain there are many other people like me who would appreciate using KDL in python. Appreciate the all your efforts! For now I am going to just use KDL in cpp. If I manage to make pykdl work then i'll post my procedure here.

MatthijsBurgh, 2021-01-22:
Should be fixed by #297; Stale issue

closed 2021-01-22 by MatthijsBurgh