orocos/orocos_kinematics_dynamics #291: Can't compile python_orocos_kdl using Python3

seanng-1, 2020-10-15:
Hi, so I am using Ubuntu 18.04 on my virtual machine and have followed the install instructions in travis.yml.

I am successfully able to compile orocos_kdl but when I try to compile python_orocos_kdl using Python3 bindings I get the following error:

[image]

I have tried different branches of the repository to test if it was a recent problem but this error continues to occur with any version that I have attempted to compile.

 I'm using sip version 4.19.7 and have installed all the following : libeigen3-dev libcppunit-dev python-sip-dev python3-sip-dev python-psutil python3-psutil python-future python3-future

I apologise for the poor formatting as this is my first time opening an issue.

MatthijsBurgh, 2020-10-15:
Just use your catkin tool of preference to build PyKDL in a catkin workspace.

seanng-1, 2020-10-20:
> Just use your catkin tool of preference to build PyKDL in a catkin workspace.

I have tried this and it still did not work.

MatthijsBurgh, 2020-10-20:
You need to update the pybind11 submodule: `git submodule update --init`

closed 2020-10-21 by seanng-1