orocos/orocos_kinematics_dynamics #292: cannot pass weight matrix to ChainIkSolverPos_LMA() class in PyKDL

radhen, 2020-10-22:
Is it possible to pass my own **weight matrix** to ChainIkSolverPos_LMA solver in PyKDL? Below is the constructer for the ChainIkSolverPos_LMA() class, 

`ChainIkSolverPos_LMA (const KDL::Chain &_chain, const Eigen::Matrix< double, 6, 1 > &_L, double _eps=1E-5, int _maxiter=500, double _eps_joints=1E-15)`

and L is the **weight matrix**. I can pass all the other arguments but not L. 
I have PyKDL installed in ROS and I am using it with Python 2.7.

MatthijsBurgh, 2020-10-23:
I am not sure if you are using a released version of KDL/PyKDL or the master branch.

But I think this applies to both. Both for the PyBind11 version(only master) and the SIP version(released and master) of PyKDL only the short constructor is available, so matching https://github.com/orocos/orocos_kinematics_dynamics/blob/master/orocos_kdl/src/chainiksolverpos_lma.hpp#L108.

A constructor like https://github.com/orocos/orocos_kinematics_dynamics/blob/master/orocos_kdl/src/chainiksolverpos_lma.hpp#L95 is not (yet) available in PyKDL.

radhen, 2020-10-23:
Good to know. Thank you :)

closed 2020-10-23 by radhen